opendilab/InterFuser #3: Lack of experiment on data efficiency / ablation study on same dataset for other methods

Kin-Zhang, 2022-09-13:
Thanks for your work.

But I still have questions on the experiment part, I notice that here:

>Dataset collection We ran a rule-based expert agent on all 8 kinds of towns and weather with 2 FPS, to collect an expert dataset of **3M frames** (410 hours). 

As I knew that in [CVPR'21 Transfuser] and [CVPR'22 LAV] dataset sizes are **all around 300-400K frames**. [1M=1,000 K]

I didn't find the result table in the paper with the same dataset training but only the online leaderboard. Maybe you have some insights that didn't point out in the paper, so I propose this issue.

This 3M=3000K with 300-400K which makes it unclear whether it's the model or the extra data that brings the boost to the performance. have you done any experiments on the data efficiency of your model?

deepcs233, 2022-09-13:
Hi, Kin-Zhang

3M frames includes 8 towns * 21 kind of weather. In our work, we only take use of part of towns and weather and the traning dataset size is about 1.5M. We will update the details in the camera ready version.

In the stage of data collection, one same route will be collected at different weather repeatly. So the dataset is redundant and we plan to use less data in the future. In our early version, we try to use about 1/5 training size and the performance only has a slight drop in our offline benchmark.

Kin-Zhang, 2022-09-13:
Thanks for your reply. I think maybe it could open this issue to remind others if they want to do some comparison on the same dataset with other open-source methods. And propose the result here.

Looking forward to your implementation.

Kait0, 2022-09-15:
Two minor details to add to this discussion: 
The CVPR TransFuser you mentioned only uses 150k frames as per their appendix. 

> In the stage of data collection, one same route will be collected at different weather repeatly. So the dataset is redundant

If you are repeating the same route with a different weather you are implicitly also randomizing the traffic situation / cars encountered because the CARLA traffic manager is not deterministic in CARLA 0.9.10.1
The thing that will be redundant is the route layout / static environment.

Kin-Zhang, 2022-09-15:
Thanks for @Kait0 pointing out here detail. And I check also the [TPAMI'22] Transfuser training data are also 228k frames in total which is also smaller compared with others. Is that the online leaderboard dataset size also?

Kait0, 2022-09-15:
yes

deepcs233, 2022-09-15:
> 

In our data collection setting, one same route will be collected 21 times (we have 21 kind of weathers). Besides  the route layout and static environment is the same, the active scenarios is also exactly the same (because we use the same route file and scenario file in each collection). Actually, there was a bug in an earlier version of the project that use only about 1/5 of the data for training. Before we fix the bug, interfuser still perform better than Transfuser[CVPR'21].
Finally, we thank your interest on our work and suggestions. We may add some experiments which use a smaller dataset in our CoRL 2022 camera ready version.

deepcs233, 2022-09-15:
> Thanks for @Kait0 pointing out here detail. And I check the [TPAMI'22] Transfuser training data are around 228k frames in total which is also smaller compared with others. Is that the online leaderboard dataset size also?

@Kait0 The online leaderboard doesn't provide a dataset. It only provides the training routes and scenarios.

deepcs233, 2022-09-15:
> 

Yes, std > 0 and is a small number. So we claim the dataset is **redundant** not the **same**.

Kait0, 2022-09-15:
Don't think the magnitude of the DS std is a meaningful metric. Different situations don't necessarily lead to different driving scores, since the model might solve both.
I can imagine that experiencing different constellations of cars in the dataset might be beneficial for training which is why I mentioned it.
I think it is an open question since it is unclear how much the difference from the carla randomness is or whether is helps at all (I think there is no paper right now that investigates this).
I would be interested if you make a dataset ablation on this for your camera ready version like you mentioned.

Kin-Zhang, 2022-09-15:
> Actually, there was a bug in an earlier version of the project that use only about 1/5 of the data for training. Before we fix the bug, interfuser still perform better than Transfuser[CVPR'21].

Sorry for the misunderstanding here. **WHAT I MEANS** here is that Interfuse didn't do a fair comparison. Like Transfuser and LAV which made the comparison with other models **in the same dataset** which holds the fair for comparison in their paper.
Control the variable like dataset holds the fair which compares others' methods in your dataset to prove it's the model and method achieve the result but not the large dataset, but I knew it's a really large workload for these.

Let's see only the data size here. Even the 1/5 data size 3000/5=600K is still larger than others like Transfuser, LAV, and TCP. But as I mentioned before, it's a really large workload to compare others' methods in your dataset with fairness.

> the active scenarios is also exactly the same (because we use the same route file and scenario file in each collection)

Even for some redundant datasets as @Kait0  mentioned. I also noticed that the event would be the same, but the TrafficManager for other NPC is still random in CARLA 0.9.10.1.

deepcs233, 2022-09-15:
Thanks, i will leave the discussions here and there are two points needed to be clarified:
1) 3000k data size is the full size of the dataset, we dont use all of them to train the agent. In training stage, we take about 6 towns and 12 weathers which probably accounted for 1500k size.
2) TCP uses the technic of model embedding, Rails uses 1M frames, IARL uses 40M frames. LAV comapres Transfuser[CVPR '21] in the paper.  However, LAV uses 400K data and Transfuser uses 150k data, which is also three times. Different methods use different camera and sensors. Totally comparasion is difficult but we can also find the contributions from these works. We will update the detailed experiment in the following version.

Kait0, 2022-09-15:
> WHAT I MEANS here is that Interfuse didn't do a fair comparison. 
You are holding them to an unreasonable standard here.
Ablations are usually done in papers with the same dataset, but system level comparisons to other papers are almost never with a comparable dataset (also for LAV and TransFuser).
Most methods differ widely in the amount of training data, the kind of training labels, sensors inputs, pre-training, training method, run time, architecture and engineering.
Progress in CARLA has come from progress on all of these things, not just the architecture.

Kin-Zhang, 2022-09-15:
> Ablations are usually done in papers with the same dataset, but system level comparisons to other papers are almost never with a comparable dataset (also for LAV and TransFuser).

This issue is for **dataset size** but not the system level and the reason you just mentioned.

deepcs233, 2022-09-16:
@Kait0 @Kin-Zhang Thanks for your disscussions. I will compelete the ablation studies about dataset size and update the results here.

Kait0, 2022-10-26:
Perhaps relevant for the people interested in this discussion.
https://arxiv.org/abs/2210.14222v1
Some of my colleagues made a ablation study on the effects of dataset scaling in the privileged setting (Table 2 b).
They recollected the TransFuser 228k dataset with different seeds 1x, 2x and 3x.
This will randomize the traffic situations encountered (but the routes/ environment is the same).
For their privileged models, scaling the dataset leads to consistent improvements in DS.

Kin-Zhang, 2022-11-01:
Screenshot from the paper <_PlanT: Explainable Planning Transformers via Object-Level Representations_> as Kait0 mentioned:

[image]

This may lead us to think more: the dataset size that 10x likes 2-3M with others' models may also achieve such high scores or even higher (maybe).

Recent NIPS work <_Model-Based Imitation Learning for Urban Driving_> also uses the 2.9M dataset size, I linked their issue also:
https://github.com/wayveai/mile/issues/4