opendilab/InterFuser #6: Model cannot be created

jdmdj1999, 2022-11-21:
Thanks for your work. I want to make an evaluation but the model cannot be created. How do I load the model？
`self.net = create_model(self.config.model)`

> RuntimeError: Unknown model (interfuser_baseline_seperate_all)

deepcs233, 2022-11-21:
Sorry, the code has a small mistake and i have fixed it.

The details:
https://github.com/opendilab/InterFuser/blob/cd0274af6b918019e9c6d76648f80a272474db32/leaderboard/team_code/interfuser_config.py#L25

jdmdj1999, 2022-11-21:
Thanks a lot

closed 2023-03-03 by PaParaZz1

Yyb-XJTU, 2023-05-15:
Could not set up the required agent:
> Unknown model (interfuser_baseline)
RuntimeError: Unknown model (interfuser_baseline)
(Thanks for your work. I want to make an evaluation but the model cannot be created. How do I load the model？)

deepcs233, 2023-05-15:
Hi,
Have you installed timm before? You could try uninstall timm and then run the below script:
`cd interfuser
python setup.py develop`

Yyb-XJTU, 2023-05-15:
Thanks a lot, but i have new problem: (The program seems to be stuck here)

> Setting up the agent
load model: leaderboard/team_code/interfuser.pth.tar
routes_town05_long_05_15_19_36_47
> Loading the world
load
load
load
load
load
load
...

Yyb-XJTU, 2023-05-15:
okk, thanks!