opendilab/InterFuser #8: Errors when running tools/data scripts and train.py

hz3014, 2022-11-24:
Thanks for the great work. However, there were some error encountered and I would be grateful if you can help with it.
error 1:
command: python tools/data/batch_merge_data.py '/media/avt/InterFuser/dataset/'
error: 
> Traceback (most recent call last):
>   File "tools/data/batch_merge_data.py", line 52, in <module>
>     for line in open(list_file, "r").readlines():
> IsADirectoryError: [Errno 21] Is a directory: '/media/avt/4TB/InterFuser/dataset/'
It seems like batch_merge_data.py needs data directory as input. however, it has strange error. What kind of input does it take?

error 2:
command: python timm/data/carla_dataset.py
error: 
> Traceback (most recent call last):
>   File "timm/data/carla_dataset.py", line 12, in <module>
>     from .base_io_dataset import BaseIODataset
> ImportError: attempted relative import with no known parent package
BTW, I did run python setup.py develop. 

error 3 is distributed training related, I changed GPU_NUM=1 since i have only one gpu.
command: bash scripts/train.sh
error: 
> Traceback (most recent call last):
>   File "train.py", line 1844, in <module>
>     main()
>   File "train.py", line 971, in main
>     args.lr * args.batch_size * torch.distributed.get_world_size() / 512.0
>   File "/home/avt/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/distributed_c10d.py", line 1067, in get_world_size
>     return _get_group_size(group)
>   File "/home/avt/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/distributed_c10d.py", line 453, in _get_group_size
>     default_pg = _get_default_group()
>   File "/home/avt/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/distributed_c10d.py", line 585, in _get_default_group
>     "Default process group has not been initialized, "
> RuntimeError: Default process group has not been initialized, please make sure to call init_process_group.
> 
torch.distributed.get_world_size() does not seems to be initialized with gpu_num= 1. Can i just replace it with value 1?

deepcs233, 2022-11-24:
HI,
**for error 1:**
batch_merge_data.py does not have to be executed, the script is used to merge small files to reduce IO preesure. The input of this script is a txt file which has the following structure:

> data/route_dir_1/
> data/route_dir_2/
> ...

**for error 2:**
_timm/data/carla_dataset.py_ can not be run directly, it is used in the training loop. There is no need to run it.

**for error 3:**
sorry i'm not clear whether pytorch distributed module supports single gpu, may be you can set args.distributed = False. But i havn't trained it on this mode, it may exist some bugs.

hz3014, 2022-11-25:
@deepcs233 Thanks for clarifying.
 I think simple args.distributed=False is not enough as other train code contains distributed setting. For now, i replaced torch.distributed.get_world_size() with value 1.
Now I am seeing another error which is asking for 'dataset/dataset_index.txt'
It seems like the same file you mentioned 

> data/route_dir_1/
> data/route_dir_2/
> ...

I have tried some route_dir but none of them works. 
weather-0/data/routes_town01_long_w0_11_16_01_18_08/
dataset/weather-0/data/routes_town01_long_w0_11_16_01_18_08/
/media/avt/4TB/InterFuser/dataset/weather-0/data/routes_town01_long_w0_11_16_01_18_08

the error code is:

> Traceback (most recent call last):
>   File "train.py", line 1858, in <module>
>     main()
>   File "train.py", line 1154, in main
>     pin_memory=args.pin_mem,
>   File "/media/avt/4TB/InterFuser/interfuser/timm/data/carla_loader.py", line 118, in create_carla_loader
>     loader = loader_class(dataset, **loader_args)
>   File "/home/avt/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 344, in __init__
>     sampler = RandomSampler(dataset, generator=generator)  # type: ignore[arg-type]
>   File "/home/avt/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/utils/data/sampler.py", line 108, in __init__
>     "value, but got num_samples={}".format(self.num_samples))
> ValueError: num_samples should be a positive integer value, but got num_samples=0

What's the correct format in dataset_index.txt? Is there a script which auto generate the txt file?

deepcs233, 2022-11-25:
The correct format in dataset_index.txt should be :

> route_path_dir `data frames in this dir`

for example:

> weather-0/data/routes_town01_long_w0_11_16_01_18_08/ 231
> weather-1/data/routes_town02_long_w0_11_16_01_13_13/ 239

231 is the number of the frames in  _weather-0/data/routes_town01_long_w0_11_16_01_18_08/rgb_ or _weather-0/data/routes_town01_long_w0_11_16_01_18_08/rgb_front_ etc.

i will update the details of dataset_index.txt, sorry for the

hz3014, 2022-12-22:
Hi@deepcs233 I have generated dataset_index.txt in the following format:

> weather-7/data/routes_town06_long_w7_11_28_18_28_35/1062
> weather-2/data/routes_town01_short_w2_11_16_08_27_10/1785
> weather-2/data/routes_town01_short_w2_11_16_09_55_05/918
> weather-2/data/routes_town02_short_w2_11_16_22_55_25/134
> weather-2/data/routes_town01_short_w2_11_16_11_44_08/569
> weather-2/data/routes_town06_short_w2_11_19_13_12_03/1168
> weather-2/data/routes_town03_long_w2_11_17_22_59_45/366

However, the code is still not working, the error is as follows. it seem to me that there is not datasets loaded. Not sure if other dataset preprocessing is needed. Would you please give some hint what might have gone wrong? much appropriated. 

> bash scripts/train.sh
> /home/avt/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/launch.py:188: FutureWarning: The module torch.distributed.launch is deprecated
> and will be removed in future. Use torchrun.
> Note that --use_env is set by default in torchrun.
> If your script expects `--local_rank` argument to be set, please
> change it to read from `os.environ['LOCAL_RANK']` instead. See 
> https://pytorch.org/docs/stable/distributed.html#launch-utility for 
> further instructions
> 
>   FutureWarning,
> WARNING:torch.distributed.run:
> *****************************************
> Setting OMP_NUM_THREADS environment variable for each process to be 1 in default, to avoid your system being overloaded, please further tune the variable for optimal performance in your application as needed. 
> *****************************************
> Added key: store_based_barrier_key:1 to store for rank: 0
> Added key: store_based_barrier_key:1 to store for rank: 1
> Rank 1: Completed store-based barrier for key:store_based_barrier_key:1 with 2 nodes.
> Training in distributed mode with multiple processes, 1 GPU per process. Process 1, total 2.
> Rank 0: Completed store-based barrier for key:store_based_barrier_key:1 with 2 nodes.
> Training in distributed mode with multiple processes, 1 GPU per process. Process 0, total 2.
> Loading pretrained weights from url (https://github.com/rwightman/pytorch-image-models/releases/download/v0.1-weights/resnet50d_ra2-464e36ba.pth)
> Loading pretrained weights from url (https://github.com/rwightman/pytorch-image-models/releases/download/v0.1-weights/resnet50d_ra2-464e36ba.pth)
> Model interfuser_baseline created, param count:52935567
> Data processing configuration for current model + dataset:
> 	input_size: (3, 224, 224)
> 	interpolation: bicubic
> 	mean: (0.485, 0.456, 0.406)
> 	std: (0.229, 0.224, 0.225)
> 	crop_pct: 0.875
> CNN backbone and transformer blocks using different learning rates!
> 165 weights in the cnn backbone, 274 weights in other modules
> AMP not enabled. Training in float32.
> Using native Torch DistributedDataParallel.
> Scheduled epochs: 35
> Sub route dir nums: 0
> Sub route dir nums: 0
> Sub route dir nums: 0
> Sub route dir nums: 0
> Current checkpoints:
>  ('./output/20221222-160204-interfuser_baseline-224-interfuser_baseline/checkpoint-0.pth.tar', 0)
> 
> Current checkpoints:
>  ('./output/20221222-160204-interfuser_baseline-224-interfuser_baseline/checkpoint-0.pth.tar', 0)
>  ('./output/20221222-160204-interfuser_baseline-224-interfuser_baseline/checkpoint-1.pth.tar', 0)
> 
> Current checkpoints:
>  ('./output/20221222-160204-interfuser_baseline-224-interfuser_baseline/checkpoint-0.pth.tar', 0)
>  ('./output/20221222-160204-interfuser_baseline-224-interfuser_baseline/checkpoint-1.pth.tar', 0)
>  ('./output/20221222-160204-interfuser_baseline-224-interfuser_baseline/checkpoint-2.pth.tar', 0)
> 
> Current checkpoints:
>  ('./output/20221222-160204-interfuser_baseline-224-interfuser_baseline/checkpoint-0.pth.tar', 0)
>  ('./output/20221222-160204-interfuser_baseline-224-interfuser_baseline/checkpoint-1.pth.tar', 0)
>  ('./output/20221222-160204-interfuser_baseline-224-interfuser_baseline/checkpoint-2.pth.tar', 0)
>  ('./output/20221222-160204-interfuser_baseline-224-interfuser_baseline/checkpoint-3.pth.tar', 0)
> 
> Current checkpoints:
>  ('./output/20221222-160204-interfuser_baseline-224-interfuser_baseline/checkpoint-0.pth.tar', 0)
>  ('./output/20221222-160204-interfuser_baseline-224-interfuser_baseline/checkpoint-1.pth.tar', 0)
>  ('./output/20221222-160204-interfuser_baseline-224-interfuser_baseline/checkpoint-2.pth.tar', 0)
>  ('./output/20221222-160204-interfuser_baseline-224-interfuser_baseline/checkpoint-3.pth.tar', 0)
>  ('./output/20221222-160204-interfuser_baseline-224-interfuser_baseline/checkpoint-4.pth.tar', 0)

deepcs233, 2022-12-22:
Hey,

> Sub route dir nums: 0
This line means that the training code couldn'd find dataset.

Can you share the running script or check the args `data_dir`. For example, /path/to/dir/root_dir/weather-7/data/routes_town06_long_w7_11_28_18_28_35/1062 needs exist. And you check that the towns and weather you specify in the args have overlapped the dataset.

hz3014, 2022-12-22:
> 

Just want to clarify that under /path/to/dir/root_dir/weather-7/data/routes_town06_long_w7_11_28_18_28_35, it follows a folder whose name is the number of frames?

for example under your current instruction, the dataset folder has following structure, is it wrongly structured?

> media/avt/4TB/InterFuser/dataset/weather-0/data/routes_town01_long_w0_11_16_01_18_08$ ls
> 2d_bbs_front  actors_data  depth_left    rgb_front  seg_front  waypoints.npy
> 2d_bbs_left   affordances  depth_right   rgb_left   seg_left
> 2d_bbs_right  birdview     lidar         rgb_rear   seg_right
> 3d_bbs        depth_front  measurements  rgb_right  topdown
>

deepcs233, 2022-12-22:
That's ok, can you share the full training script?

hz3014, 2022-12-22:
> That's ok, can you share the full training script?

I use bash scripts/train.sh to train. The train.py is same as yours.
under train.sh is  

> GPU_NUM=2
> DATASET_ROOT='/media/avt/4TB/InterFuser/dataset/'
> 
> ./distributed_train.sh $GPU_NUM $DATASET_ROOT  --dataset carla --train-towns 1 2 3 4 6 7 10  --val-towns 5 \
>     --train-weathers 0 1 2 3 4 5 6 7 8 9  --val-weathers 10 11 12 13 \
>     --model interfuser_baseline --sched cosine --epochs 25 --warmup-epochs 5 --lr 0.0005 --batch-size 16  -j 16 --no-prefetcher --eval-metric l1_error \
>     --opt adamw --opt-eps 1e-8 --weight-decay 0.05  \
>     --scale 0.9 1.1 --saver-decreasing --clip-grad 10 --freeze-num -1 \
>     --with-backbone-lr --backbone-lr 0.0002 \
>     --multi-view --with-lidar --multi-view-input-size 3 128 128 \
>     --experiment interfuser_baseline \
>     --pretrained

under distributed_train.sh

hz3014, 2022-12-22:
> > weather-7/data/routes_town06_long_w7_11_28_18_28_35/1062
> 
> Here should be weather-7/data/routes_town06_long_w7_11_28_18_28_35/ 1062 Between the path and frames there is a space, can you try again?

Now it is training : )

deepcs233, 2022-12-22:
Sorry for not making it clear before and thank you for your feedback.

hz3014, 2022-12-25:
Hi @deepcs233 There is another error during training which relates to the testing/validating part of the code. Appreciate if you can take a look as im not very familiar with the code yet.

> Train: 0 [25136/25137 (100%)]  Loss(traffic):  0.120556 (0.1629)  Loss(waypoints):  0.053456 (0.0654)  Loss(junction):  0.081597 (0.1816)  Loss(light):  0.011085 (0.0771)  Loss:  0.083545 (0.1195)  Time: 0.144s,  221.91/s  (0.188s,  170.10/s)  LR: 5.000e-06  Data: 0.003 (0.003)
> Test: [   0/1134]  Time: 1.120 (1.120)  Loss:  0.1938 (0.1938)  Loss(traffic):  0.1394 (0.1394)  Loss(waypoints):  0.0186 (0.0186)  Loss(junction):  2.204867 (2.2049)  Loss(light):  0.000233 (0.0002)  Acc(junction): 53.125000 (53.1250)  Acc(light): 100.000000 (100.0000)  
> Traceback (most recent call last):
>   File "train.py", line 1844, in <module>
>     main()
>   File "train.py", line 1256, in main
>     amp_autocast=amp_autocast,
>   File "train.py", line 1680, in validate
>     loss_velocity.data, args.world_size
> AttributeError: 'int' object has no attribute 'data'
> Traceback (most recent call last):
>   File "train.py", line 1844, in <module>
>     main()
>   File "train.py", line 1256, in main
>     amp_autocast=amp_autocast,
>   File "train.py", line 1680, in validate
>     loss_velocity.data, args.world_size
> AttributeError: 'int' object has no attribute 'data'
> ERROR:torch.distributed.elastic.multiprocessing.api:failed (exitcode: 1) local_rank: 0 (pid: 21088) of binary: /home/avt/anaconda3/envs/interfuser/bin/python3
> Traceback (most recent call last):
>   File "/home/avt/anaconda3/envs/interfuser/lib/python3.7/runpy.py", line 193, in _run_module_as_main
>     "__main__", mod_spec)
>   File "/home/avt/anaconda3/envs/interfuser/lib/python3.7/runpy.py", line 85, in _run_code
>     exec(code, run_globals)
>   File "/home/avt/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/launch.py", line 195, in <module>
>     main()
>   File "/home/avt/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/launch.py", line 191, in main
>     launch(args)
>   File "/home/avt/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/launch.py", line 176, in launch
>     run(args)
>   File "/home/avt/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/run.py", line 756, in run
>     )(*cmd_args)
>   File "/home/avt/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/launcher/api.py", line 132, in __call__
>     return launch_agent(self._config, self._entrypoint, list(args))
>   File "/home/avt/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/launcher/api.py", line 248, in launch_agent
>     failures=result.failures,
> torch.distributed.elastic.multiprocessing.errors.ChildFailedError: 
> ============================================================
> train.py FAILED
> ------------------------------------------------------------
> Failures:
> [1]:
>   time      : 2022-12-25_11:49:41
>   host      : avt
>   rank      : 1 (local_rank: 1)
>   exitcode  : 1 (pid: 21089)
>   error_file: <N/A>
>   traceback : To enable traceback see: https://pytorch.org/docs/stable/elastic/errors.html
> ------------------------------------------------------------
> Root Cause (first observed failure):
> [0]:
>   time      : 2022-12-25_11:49:41
>   host      : avt
>   rank      : 0 (local_rank: 0)
>   exitcode  : 1 (pid: 21088)
>   error_file: <N/A>
>   traceback : To enable traceback see: https://pytorch.org/docs/stable/elastic/errors.html
>

deepcs233, 2022-12-25:
It looks like loss_velocity is a int, maybe you could use this code in line 1680 of train.py:

> loss_velocity, args.world_size

hz3014, 2022-12-25:
> 

is it supposed to be integer type? as following tensor operation will have further error: 
```

Traceback (most recent call last):
  File "train.py", line 1844, in <module>
    main()
  File "train.py", line 1256, in main
    amp_autocast=amp_autocast,
  File "train.py", line 1680, in validate
    loss_velocity, args.world_size
  File "/media/avt/2TB_SSD/Safey-Guaranteed-Interfuser/interfuser/timm/utils/distributed.py", line 12, in reduce_tensor
Traceback (most recent call last):
  File "train.py", line 1844, in <module>
    rt = tensor.clone()
AttributeError: 'int' object has no attribute 'clone'
    main()
  File "train.py", line 1256, in main
    amp_autocast=amp_autocast,
  File "train.py", line 1680, in validate
    loss_velocity, args.world_size
  File "/media/avt/2TB_SSD/Safey-Guaranteed-Interfuser/interfuser/timm/utils/distributed.py", line 12, in reduce_tensor
    rt = tensor.clone()
AttributeError: 'int' object has no attribute 'clone'
```

deepcs233, 2022-12-25:
https://github.com/opendilab/InterFuser/blob/f3b0316d290059349dcaa79e193f596787ebeca9/interfuser/train.py#L834 I find the reason here, maybe you can replace it with torch.tensor(0.0).cuda() ?
The evaluation part can be skipped and you can directly use the last epoch's model.

hz3014, 2023-01-09:
Hi @deepcs233, I am having some problem with Evaluation part of Interfuser.
I was wondering what is results/interfuser_result.json? and how shall we obtain the file?

deepcs233, 2023-01-09:
Hi,

It's not a bug. The code terminated because it loaded the result file and found the evaluation has completed. In the default setting the arg "RESUME" is True, so it terminated.  You can delete the file or move the file or change the save path or set the "RESUME" False to solve the problem.

The path of result file can be found in the running script, such as ./leaderboard/scripts/run_evaluation.sh. The result file records detailed experimental performance, including the overall metric (driving score, route completion, etc,.) and the performace in each route. It's the file generated by the evaluation code. Before the evaluation, we don't need to generate it by ourselves.

closed 2023-02-03 by hz3014