opendilab/InterFuser #10: Porting InterFuser to BeamNG.tech

Stoneymon, 2022-12-14:
Dear All,

I am a student working on a project involving testing ADAS/AV and scenario synthesis using [BeamNG.tech](https://beamng.tech/) and would love to run (test) your driving agent in that simulator. I know CARLA is kind of a de facto standard, but IMHO BeamNG.tech is superior when it comes to physic simulation, content, and flexibility. Further, BeamNG.tech is free for research, offers a python API, just like CARLA, and implements a wide range of sensors.

So I wonder how technically difficult it would be to port InterFuser to BeamNG.tech and whether anyone of you could support me (and my colleagues) in doing so. Hope to hear from you soon,

Thanks!

-- Benedikt Steininger

deepcs233, 2022-12-15:
Hey,

Sorry, I'm not familar with this simulator. But if this simulator supports the LiDAR sensor, multi RGB sensors and can obtain the other objects' state, the simulator migration may be easy to make. Maybe only need to modify the code about data collection and the agent, and the code about training procedure don't need to be updated.
Feel free to concat me via email or this GitHub Issue if you counter some problems when you port InterFuser to BeamNG.tech.

Thanks!

closed 2023-02-10 by deepcs233

CoderXuans, 2024-09-20:
@Stoneymon 
Hello, I have a similar question for you. Have you successfully migrated? If so, could you tell me which parts of the InterFuser code you modified? I would greatly appreciate it.