opendilab/InterFuser #11: training with one GPU, error

mmahdavian, 2022-12-30:
Hello

Thank you for this repo. I am trying to train the model with one GPU and I am receiving this error. Do you know how to fix it?

```
/home/mohammad/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/launch.py:188: FutureWarning: The module torch.distributed.launch is deprecated
and will be removed in future. Use torchrun.
Note that --use_env is set by default in torchrun.
If your script expects `--local_rank` argument to be set, please
change it to read from `os.environ['LOCAL_RANK']` instead. See 
https://pytorch.org/docs/stable/distributed.html#launch-utility for 
further instructions

  FutureWarning,
Training with a single process on 1 GPUs.
Loading pretrained weights from url (https://github.com/rwightman/pytorch-image-models/releases/download/v0.1-weights/resnet50d_ra2-464e36ba.pth)
Model interfuser_baseline created, param count:52935567
Data processing configuration for current model + dataset:
	input_size: (3, 224, 224)
	interpolation: bicubic
	mean: (0.485, 0.456, 0.406)
	std: (0.229, 0.224, 0.225)
	crop_pct: 0.875
Traceback (most recent call last):
  File "train.py", line 1847, in <module>
    main()
  File "train.py", line 972, in main
    print(torch.distributed.get_world_size())
  File "/home/mohammad/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/distributed_c10d.py", line 1067, in get_world_size
    return _get_group_size(group)
  File "/home/mohammad/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/distributed_c10d.py", line 453, in _get_group_size
    default_pg = _get_default_group()
  File "/home/mohammad/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/distributed_c10d.py", line 585, in _get_default_group
    "Default process group has not been initialized, "
RuntimeError: Default process group has not been initialized, please make sure to call init_process_group.
ERROR:torch.distributed.elastic.multiprocessing.api:failed (exitcode: 1) local_rank: 0 (pid: 120559) of binary: /home/mohammad/anaconda3/envs/interfuser/bin/python3
Traceback (most recent call last):
  File "/home/mohammad/anaconda3/envs/interfuser/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/mohammad/anaconda3/envs/interfuser/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/mohammad/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/launch.py", line 195, in <module>
    main()
  File "/home/mohammad/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/launch.py", line 191, in main
    launch(args)
  File "/home/mohammad/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/launch.py", line 176, in launch
    run(args)
  File "/home/mohammad/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/run.py", line 756, in run
    )(*cmd_args)
  File "/home/mohammad/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/launcher/api.py", line 132, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
  File "/home/mohammad/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/launcher/api.py", line 248, in launch_agent
    failures=result.failures,
torch.distributed.elastic.multiprocessing.errors.ChildFailedError: 
============================================================
train.py FAILED
------------------------------------------------------------
Failures:
  <NO_OTHER_FAILURES>
------------------------------------------------------------
Root Cause (first observed failure):
[0]:
  time      : 2022-12-29_16:44:22
  host      : mohammad
  rank      : 0 (local_rank: 0)
  exitcode  : 1 (pid: 120559)
  error_file: <N/A>
  traceback : To enable traceback see: https://pytorch.org/docs/stable/elastic/errors.html

```

deepcs233, 2022-12-30:
Hey, maybe you can refer to https://github.com/opendilab/InterFuser/issues/8#issuecomment-1326742076, set args.distributed=False

mmahdavian, 2022-12-30:
Thanks. Setting "torch.distributed.get_world_size()" to 1 solved the problem.

closed 2022-12-30 by mmahdavian