opendilab/InterFuser #12: Long,short or tiny

mmahdavian, 2023-01-01:
Hello

I was wondering which length is better suited for training the models? Long, short or tiny bash files? Which was did you use for training your model?

Thank You

deepcs233, 2023-01-01:
Hi
If you have enough disk space, i recommand to collect balanced data with different length, such as run town01_long and ,town02_short, town03_tiny. The dataset with shorter length includes more data around the junction and the dataset with longer length includes more data about the scenarios happened  on the road.

mmahdavian, 2023-01-03:
Thanks for your suggestion. I have some questions regarding the dataset.

1-While generating the dataset sometimes I see comments like this:

```
========= Results of RouteScenario_64 (repetition 0) ------ FAILURE =========

╒═════════════════════════════════╤═════════════════════╕
│ Start Time                      │ 2023-01-03 14:34:30 │
├─────────────────────────────────┼─────────────────────┤
│ End Time                        │ 2023-01-03 14:37:13 │
├─────────────────────────────────┼─────────────────────┤
│ Duration (System Time)          │ 163.63s             │
├─────────────────────────────────┼─────────────────────┤
│ Duration (Game Time)            │ 29.05s              │
├─────────────────────────────────┼─────────────────────┤
│ Ratio (System Time / Game Time) │ 0.178               │
╘═════════════════════════════════╧═════════════════════╛

╒═══════════════════════╤═════════╤═════════╕
│ Criterion             │ Result  │ Value   │
├───────────────────────┼─────────┼─────────┤
│ RouteCompletionTest   │ FAILURE │ 16.75 % │
├───────────────────────┼─────────┼─────────┤
│ OutsideRouteLanesTest │ SUCCESS │ 0 %     │
├───────────────────────┼─────────┼─────────┤
│ CollisionTest         │ SUCCESS │ 0 times │
├───────────────────────┼─────────┼─────────┤
│ RunningRedLightTest   │ SUCCESS │ 0 times │
├───────────────────────┼─────────┼─────────┤
│ RunningStopTest       │ SUCCESS │ 0 times │
├───────────────────────┼─────────┼─────────┤
│ InRouteTest           │ SUCCESS │         │
├───────────────────────┼─────────┼─────────┤
│ AgentBlockedTest      │ SUCCESS │         │
├───────────────────────┼─────────┼─────────┤
│ Timeout               │ FAILURE │         │
╘═══════════════════════╧═════════╧═════════╛

```
Is this OK? 

2- Also, it seems like the car sometimes stops in the middle of the road and I see comments like this:
```
destory hazard actor: id: 13640, type: vehicle.diamondback.century
destory hazard actor: id: 13647, type: vehicle.diamondback.century
destory hazard actor: id: 13674, type: vehicle.diamondback.century
destory hazard actor: id: 13706, type: vehicle.audi.tt
destory hazard actor: id: 13766, type: vehicle.dodge_charger.police
destory hazard actor: id: 13768, type: vehicle.diamondback.century
destory hazard actor: id: 13728, type: vehicle.audi.tt
```

3- In general based on "routes_town01_tiny.xml" file there are 291 routes for it. Do we need all of them? The dataset generator sometime stops and doesn't generate all of them. Also, some of them are similar based of what I see from rgb_front images. Is this natural?

deepcs233, 2023-01-04:
Hi
1. It's OK if this type of "failure" log didn't occur frequently.  The avg driving score should be greater than 0.75 after you run all cases in each .xml file. You can check the metric in the result.json.
2. It's OK because we delete some blocked vehicles to help the ego-car continue to drive along the route.
3. (1) It's better to run all of them, the length of tiny route is very short and it doesn't cost much time. (2) About the a abnormal stop of generating dataset, could provide more log? It can be re-runed (3) Yes, they may look similiar based on the rgb_front images, but the route of them differs. For example, in the same crossing where one route may turn left but the other route may turn right. It can collect more comprehensive and diverse dataset. In short, if you have enough disk space, i recommand you to run all of them. If not, you can only collect a subset of them which may lead to a alight performance drop.

mmahdavian, 2023-01-05:
Thank you so much

mmahdavian, 2023-01-05:
By the way is it possible to run the simulators without graphical rendering for making it faster? Would it be able to generate the rgb_images?

closed 2023-01-05 by mmahdavian

deepcs233, 2023-01-05:
Actually, we collected the dataset on the gpu server which only has terminals, we ran the simulators without graphical rendering. We found it can collect complete and correct data. But we didn't find any speed up compared with the computers with graphical interface.

mmahdavian, 2023-01-05:
OK. Thanks.