opendilab/InterFuser #13: town 6 doesn't work

mmahdavian, 2023-01-05:
Hi. I am trying to generate dataset from town 6 and it doesn't work.I run "bash data_collection/batch_run/run_route_routes_town06_short.sh" and this what I am seeing:
```

leaderboard/leaderboard/leaderboard_evaluator.py:91: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(dist.version) < LooseVersion('0.9.10'):

========= Preparing RouteScenario_0 (repetition 0) =========
> Setting up the agent
{'destory_hazard_actors': True, 'rgb_only': False, 'save_skip_frames': 10, 'waypoint_disturb': 0.2, 'waypoint_disturb_seed': 2020, 'weather': 0}
10
routes_town06_short_w0_01_04_16_03_36
> Loading the world
load
```

After that it stops and the carla simulator gives the following error:

[image]

Do you know what the reason is? (By the way I have tested it with different PCs and get the same result. The other towns are OK!)

deepcs233, 2023-01-05:
HI
I don't find the error log in your post, "Carla is not responding" isn't a error, you could wait for a few minutes because the scenarios in some maps may be very much and need more time to load them. 
By the way, have you downloaded the extra town maps?

mmahdavian, 2023-01-19:
Thank you. I am not sure about extra town maps. I have town 1 to 7 I think. Is there any other maps to download?

deepcs233, 2023-01-19:
In addition to these map, Town10 is also included. I think if you ran the script setup_carla.sh, you have already downloaded the all maps.