opendilab/InterFuser #16: Generating Data on GPU servers

mmahdavian, 2023-01-20:
Hello

You previously mentioned you have tried to generate the dataset on headless machines. I am trying to do the same on a GPU server and I have tried following commands: 
```
CUDA_VISIBLE_DEVICES=0 ./CarlaUE4.sh --world-port=20000 -opengl &
CUDA_VISIBLE_DEVICES=0 ./CarlaUE4.sh --world-port=20000 -opengl -carla-server &
DISPLAY= ./CarlaUE4.sh --world-port=20000 -opengl -carla-server &
```
But non of them works and I get following error. Do you know what the problem is? Thank You
```
Disabling core dumps.
error: XDG_RUNTIME_DIR not set in the environment.
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM default
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM default
terminating with uncaught exception of type std::__1::system_error: thread constructor failed: Resource temporarily unavailable
Signal 6 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=6
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=124960 LargeMemoryPoolOffset=256096 
Engine crash handling finished; re-raising signal 6 for the default handler. Good bye.
```

deepcs233, 2023-01-21:
I have experienced this problem, and it is the cause of the program itself. 
I recommend you:
1. change a gpu with a larger memory
2. do not use the flag --opengl, like `CUDA_VISIBLE_DEVICES=0 ./CarlaUE4.sh --world-port=20000`
3. run fewer simulators in one gpu card
4. re-run the simulator and continue to collect the data when it crashed

deepcs233, 2023-01-24:
https://github.com/carla-simulator/carla/issues/3540#issuecomment-1401669518