opendilab/InterFuser #18: Data generation in different weathers

mmahdavian, 2023-02-06:
Hi. I am trying to generate data in different weathers. For example 3 to 5. And I edit the data_collection/batch_run/run_route_routes_town01_long.sh to:
```
bash data_collection/bashs/weather-3/routes_town02_long.sh &
bash data_collection/bashs/weather-4/routes_town02_long.sh &
bash data_collection/bashs/weather-5/routes_town02_long.sh &
```
But it doesn't start to generate the dataset.But if it starts from weather-0,1,2 and then 3 to 5 it works. How should I change it to other weathers, so I can generate the dataset from different weathers randomly?

Thank You

deepcs233, 2023-02-06:
Hi, can you provide the log or the output? Besides, why the script `run_route_routes_town01_long` collected the data in town02, rather than town01.

mmahdavian, 2023-02-07:
Yes. Sorry both are town01. I edited them. What I see after running the bash file is the following, after almost 5 to 10 minutes of doing nothing:

`Traceback (most recent call last):
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 479, in main
    leaderboard_evaluator = LeaderboardEvaluator(arguments, statistics_manager)
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 479, in main
    leaderboard_evaluator = LeaderboardEvaluator(arguments, statistics_manager)
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 479, in main
    leaderboard_evaluator = LeaderboardEvaluator(arguments, statistics_manager)
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 87, in __init__
    self.traffic_manager = self.client.get_trafficmanager(int(args.trafficManagerPort))
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 87, in __init__
    self.traffic_manager = self.client.get_trafficmanager(int(args.trafficManagerPort))
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 87, in __init__
    self.traffic_manager = self.client.get_trafficmanager(int(args.trafficManagerPort))
RuntimeError: time-out of 600000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:20008
RuntimeError: time-out of 600000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:20014
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 479, in main
    leaderboard_evaluator = LeaderboardEvaluator(arguments, statistics_manager)
RuntimeError: time-out of 600000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:20012
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 87, in __init__
    self.traffic_manager = self.client.get_trafficmanager(int(args.trafficManagerPort))
Exception ignored in: <bound method LeaderboardEvaluator.__del__ of <__main__.LeaderboardEvaluator object at 0x7fdcc1146c88>>
Exception ignored in: <bound method LeaderboardEvaluator.__del__ of <__main__.LeaderboardEvaluator object at 0x7f3e65fd0c88>>
Traceback (most recent call last):
RuntimeError: time-out of 600000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:20010
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 124, in __del__
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 124, in __del__
Exception ignored in: <bound method LeaderboardEvaluator.__del__ of <__main__.LeaderboardEvaluator object at 0x7f6c623e1c88>>
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 124, in __del__
Exception ignored in: <bound method LeaderboardEvaluator.__del__ of <__main__.LeaderboardEvaluator object at 0x7f5cfc708c88>>
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 124, in __del__
    self._cleanup()
    self._cleanup()
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 136, in _cleanup
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 136, in _cleanup
    self._cleanup()
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 136, in _cleanup
    self._cleanup()
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 136, in _cleanup
    if self.manager and self.manager.get_running_status() \
AttributeError: 'LeaderboardEvaluator' object has no attribute 'manager'
    if self.manager and self.manager.get_running_status() \
AttributeError: 'LeaderboardEvaluator' object has no attribute 'manager'
    if self.manager and self.manager.get_running_status() \
    if self.manager and self.manager.get_running_status() \
AttributeError: 'LeaderboardEvaluator' object has no attribute 'manager'
AttributeError: 'LeaderboardEvaluator' object has no attribute 'manager'
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 489, in <module>
    main()
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 485, in main
    del leaderboard_evaluator
UnboundLocalError: local variable 'leaderboard_evaluator' referenced before assignment
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 489, in <module>
    main()
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 485, in main
    del leaderboard_evaluator
UnboundLocalError: local variable 'leaderboard_evaluator' referenced before assignment
Traceback (most recent call last):
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 489, in <module>
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 489, in <module>
    main()
    main()
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 485, in main
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 485, in main
    del leaderboard_evaluator
    del leaderboard_evaluator
UnboundLocalError: local variable 'leaderboard_evaluator' referenced before assignment
UnboundLocalError: local variable 'leaderboard_evaluator' referenced before assignment
`

deepcs233, 2023-02-07:
It looks like the CARLA simulator crashed (I often encounter this problem). I recommend you kill all CARLA processes, and restart them, then re-run the script.
And, different weather data collector based on different ports which runs the CARLA. For example, weather-3,4,5 needs the CARLA runs on the port 20008, 20010, 20012. Please make sure there are three CARLA runs on these ports.
Lastly, if you just want to collect the data with random weather, you can replace the code https://github.com/opendilab/InterFuser/blob/f3b0316d290059349dcaa79e193f596787ebeca9/leaderboard/team_code/base_agent.py#L62 with 
`self.weather_id = random.randint(0,20)`. Then you could run the script data_collection/bashs/weather-1/routes_town01_long.sh to collect data with random weather. The full weather list can refer to https://github.com/opendilab/InterFuser/blob/f3b0316d290059349dcaa79e193f596787ebeca9/leaderboard/team_code/auto_pilot.py#L24

mmahdavian, 2023-02-07:
Thanks. I didn't know about the port numbers.

closed 2023-02-07 by mmahdavian