opendilab/InterFuser #20: ValueError: num_samples should be a positive integer value, but got num_samples=0

xanhug, 2023-02-23:
Thank you for your amazing work!
I have encountered some error like https://github.com/opendilab/InterFuser/issues/8, I'm not sure if there are something wrong in my data generation process or my "dataset_index.txt",could you please give me some hints?

here is the log when i run "bash scripts/train.sh"

``` 
(interfuser) hxa@hxa-Nitro-AN515-58:~/InterFuser/interfuser$ bash scripts/train.sh
/home/hxa/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/launch.py:188: FutureWarning: The module torch.distributed.launch is deprecated
and will be removed in future. Use torchrun.
Note that --use_env is set by default in torchrun.
If your script expects `--local_rank` argument to be set, please
change it to read from `os.environ['LOCAL_RANK']` instead. See 
https://pytorch.org/docs/stable/distributed.html#launch-utility for 
further instructions

  FutureWarning,
Training with a single process on 1 GPUs.
Loading pretrained weights from url (https://github.com/rwightman/pytorch-image-models/releases/download/v0.1-weights/resnet50d_ra2-464e36ba.pth)
Model interfuser_baseline created, param count:52935567
Data processing configuration for current model + dataset:
	input_size: (3, 224, 224)
	interpolation: bicubic
	mean: (0.485, 0.456, 0.406)
	std: (0.229, 0.224, 0.225)
	crop_pct: 0.875
CNN backbone and transformer blocks using different learning rates!
165 weights in the cnn backbone, 274 weights in other modules
AMP not enabled. Training in float32.
Scheduled epochs: 35
Sub route dir nums: 0
Sub route dir nums: 0
Traceback (most recent call last):
  File "train.py", line 1844, in <module>
    main()
  File "train.py", line 1140, in main
    pin_memory=args.pin_mem,
  File "/home/hxa/InterFuser/interfuser/timm/data/carla_loader.py", line 118, in create_carla_loader
    loader = loader_class(dataset, **loader_args)
  File "/home/hxa/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 344, in __init__
    sampler = RandomSampler(dataset, generator=generator)  # type: ignore[arg-type]
  File "/home/hxa/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/utils/data/sampler.py", line 108, in __init__
    "value, but got num_samples={}".format(self.num_samples))
ValueError: num_samples should be a positive integer value, but got num_samples=0
ERROR:torch.distributed.elastic.multiprocessing.api:failed (exitcode: 1) local_rank: 0 (pid: 7489) of binary: /home/hxa/anaconda3/envs/interfuser/bin/python3
Traceback (most recent call last):
  File "/home/hxa/anaconda3/envs/interfuser/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/hxa/anaconda3/envs/interfuser/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/hxa/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/launch.py", line 195, in <module>
    main()
  File "/home/hxa/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/launch.py", line 191, in main
    launch(args)
  File "/home/hxa/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/launch.py", line 176, in launch
    run(args)
  File "/home/hxa/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/run.py", line 756, in run
    )(*cmd_args)
  File "/home/hxa/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/launcher/api.py", line 132, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
  File "/home/hxa/anaconda3/envs/interfuser/lib/python3.7/site-packages/torch/distributed/launcher/api.py", line 248, in launch_agent
    failures=result.failures,
torch.distributed.elastic.multiprocessing.errors.ChildFailedError: 
============================================================
train.py FAILED
------------------------------------------------------------
Failures:
  <NO_OTHER_FAILURES>
------------------------------------------------------------
Root Cause (first observed failure):
[0]:
  time      : 2023-02-23_10:39:25
  host      : hxa-Nitro-AN515-58
  rank      : 0 (local_rank: 0)
  exitcode  : 1 (pid: 7489)
  error_file: <N/A>
  traceback : To enable traceback see: https://pytorch.org/docs/stable/elastic/errors.html
============================================================
``` 
here is my train.sh
``` 
GPU_NUM=1
DATASET_ROOT='/home/hxa/InterFuser/dataset/'

./distributed_train.sh $GPU_NUM $DATASET_ROOT  --dataset carla --train-towns 1 2 3 4 6 7 10  --val-towns 5 \
    --train-weathers 0 1 2 3 4 5 6 7 8 9  --val-weathers 10 11 12 13 \
    --model interfuser_baseline --sched cosine --epochs 25 --warmup-epochs 5 --lr 0.0005 --batch-size 16  -j 16 --no-prefetcher --eval-metric l1_error \
    --opt adamw --opt-eps 1e-8 --weight-decay 0.05  \
    --scale 0.9 1.1 --saver-decreasing --clip-grad 10 --freeze-num -1 \
    --with-backbone-lr --backbone-lr 0.0002 \
    --multi-view --with-lidar --multi-view-input-size 3 128 128 \
    --experiment interfuser_baseline \
    --pretrained
``` 
and my "dataset_index.txt" is like
``` 
/home/hxa/InterFuser/data/eval/routes_town05_long_02_22_10_54_29/ 940
``` 

I have noticed that my path of generated dataset is 
``` 
/home/hxa/InterFuser/data/eval/routes_town05_long_02_22_10_54_29/
``` 
not like
``` 
weather-7/data/routes_town06_long_w7_11_28_18_28_35/
``` 
my path haven't prefix like "weather-7" so I'm not sure if there are something wrong in my dataset generation.

xanhug, 2023-02-23:
and i found many of these folders are empty,except these folders with prefix "rgb" and "actors_data"  "birdview" "measurement",is it normal?
[image]

xanhug, 2023-02-23:
after setup the enviroment,i directly tried to generate dataset with a single CARLA Server by run 
```
carla/CarlaUE4.sh --world-port=2000 -opengl
./leaderboard/scripts/run_evaluation.sh
```
and here is my "run_evaluation.sh"
```
#!/bin/bash

export CARLA_ROOT=carla
export CARLA_SERVER=${CARLA_ROOT}/CarlaUE4.sh
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla
export PYTHONPATH=$PYTHONPATH:$CARLA_ROOT/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg
export PYTHONPATH=$PYTHONPATH:leaderboard
export PYTHONPATH=$PYTHONPATH:leaderboard/team_code
export PYTHONPATH=$PYTHONPATH:scenario_runner

export LEADERBOARD_ROOT=leaderboard
export CHALLENGE_TRACK_CODENAME=SENSORS
export PORT=2000 # same as the carla server port
export TM_PORT=2500 # port for traffic manager, required when spawning multiple servers/clients
export DEBUG_CHALLENGE=0
export REPETITIONS=1 # multiple evaluation runs
export ROUTES=leaderboard/data/training_routes/routes_town05_long.xml
export TEAM_AGENT=leaderboard/team_code/auto_pilot.py  #leaderboard/team_code/interfuser_agent.py # agent
#export TEAM_CONFIG=leaderboard/team_code/interfuser_config.py # model checkpoint, not required for expert
export CHECKPOINT_ENDPOINT=results/sample_result.json # results file
export SCENARIOS=leaderboard/data/scenarios/town05_all_scenarios.json
export SAVE_PATH=data/eval # path for saving episodes while evaluating
export RESUME=True

python3 ${LEADERBOARD_ROOT}/leaderboard/leaderboard_evaluator.py \
--scenarios=${SCENARIOS}  \
--routes=${ROUTES} \
--repetitions=${REPETITIONS} \
--track=${CHALLENGE_TRACK_CODENAME} \
--checkpoint=${CHECKPOINT_ENDPOINT} \
--agent=${TEAM_AGENT} \
--agent-config=${TEAM_CONFIG} \
--debug=${DEBUG_CHALLENGE} \
--record=${RECORD_PATH} \
--resume=${RESUME} \
--port=${PORT} \
--trafficManagerPort=${TM_PORT}

```

xanhug, 2023-02-23:
i try to still Generate Data  with a single CARLA Server but via run
```
cd carla
CUDA_VISIBLE_DEVICES=0 ./CarlaUE4.sh --world-port=20000 -opengl &
```

```
cd dataset
python init_dir.py
cd ..
cd data_collection
python generate_yamls.py # You can modify fps, waypoints distribution strength ...

# If you don't need all weather, you can modify the following script
python generate_bashs.py
python generate_batch_collect.py 
cd ..
```
and
```
bash data_collection/batch_run/run_route_routes_town01_long.sh
```
[image]

now, the generated dataset seems to be normal and no empty folders

deepcs233, 2023-02-23:
Hi, thanks for your feekback!
For Issue 1 (https://github.com/opendilab/InterFuser/issues/20#issue-1596135703), the issue is because the training progamme can not find data. The path listed in dataset_index.txt should be relative path, not absolute path. When it loads the data, it will concat the `DATASET_ROOT` and the paths listed in dataset_index.txt. Besides, make sure dataset_index.txt  is placed in directory `DATASET_ROOT`.
For Issue 2, it looks like it's running properly, so feel free to keep discussing if you have any further questions.

xanhug, 2023-02-23:
@deepcs233 Thanks for your reply!
I want to have a quick pass through this great project to improve myself,but i have not enough computing resource,the scenarios and weather conditions are too many for me,i'm not sure if i could only generate a part of dataset to train a agent?Will this cause any problems?could you please give me some advices?

deepcs233, 2023-02-23:
To reduce space usuage, i recommand you to use random weather to collect the data of all routes, rather than use 21 kinds of weather to collect the data repeatly, which will collect 21X times data. This way can cover as much weather and routes as possible with the minimal space usage. I estimate it will only use about 150G disk space and can get the best model with limited space. You can refer to https://github.com/opendilab/InterFuser/issues/18#issuecomment-1420344529 to modify the related code.

xanhug, 2023-02-23:
@deepcs233 thanks a lot ,and sorry for take up your time.

closed 2023-02-24 by xanhug