opendilab/InterFuser #24: Some issues about data size and loss in training

tulerfeng, 2023-03-06:
Hello,

Thanks for your interesting and great work. I have got some problems in training.

I uniformly collected around 300,000 frames from town01-short, town02-tiny and town03-long in 21 kinds of weathers. Then, I used these data to train Interfuser and obtain a final loss around 0.03. 

However, when I evaluated the performance of the trained model in Town05-Long, the model got bad performance as shown in the following figure (please ignore the [system time/game time] since our server is busy these days).

[image]

I want to ask the following questions:
1.	I only collect 1/10 dataset compared to your original paper. Is this bad performance caused by the small size of my collected datasets? Or are there some other reasons?
2.	How much is the final loss you can get when using 3,000,000 frames for training as in the original paper? Is my obtained loss reasonable?
3.	Our GPU-resource is limited. Could I tune up the learning rate and decade the number of epochs to accelerate training? Is the warm up epochs important? Could I decrease the number of warm up epochs?

I'm looking forward for your reply and thanks again for your great work !

deepcs233, 2023-03-06:
Hi!
1. Town01 & 02 are small maps where the road only have a single lane. If you don't want to collect too much data, i recommand to collect the data in the more maps (like 03 04 06 07 10). Also, I think 1/10 size of the dataset may be too small.
2. 0.03 is the training loss？ Using it as the metric may meaningless.I suggest that set the Town05 dataset as the vaildation dataset. From my memory (which may not be very accurate), the val loss is about 0.1-0.3.
3. Warm-up epochs is important, you could cut the other epochs to 20. If you collect more data, you could use less epochs for training, which may not speed too much time with more data :) To reduce the GPU and other resource usuage, you cold also try to reduce the layers nad feature_dim in the transformer encoders and decoders. For example, use 3 layers for the encoder and decoder, 192 for feature_dim.

tulerfeng, 2023-03-06:
Thanks for your fast reply! My questions have been addressed!

tulerfeng, 2023-03-08:
By the way, I find that there are many duplicate training routes. For example, in [leaderboard/data/training_routes/routes_town01_short.xml](https://github.com/opendilab/InterFuser/blob/main/leaderboard/data/training_routes/routes_town01_short.xml), route with id '1' and  route with id '2' are totally the same.  I haved checked my collected data and noticed that these duplicate training routes have caused me collecting many duplicate training data.

I have got some questions. Is it reasonable to collect such duplicate training data? Do you filtering out these duplicate training routes when collecting training data?

deepcs233, 2023-03-08:
Hi, if you would not like to collect much data, you may remove them. But even though the routes are totally same,the background objects (traffic lights, vehichles, pedestrians) are random, and are different for each run. I didn't filter out these 'duplicate' training data when i use them.

tulerfeng, 2023-03-08:
OK，thanks for your reply!

closed 2023-05-06 by tulerfeng