opendilab/InterFuser #26: How to run train.py on single GPU PC？

FallenMeteorite, 2023-03-13:
Because of my finite conditions, I have only one GPU which can train it. I modified train.sh GPU_NUM=1 but there is an error: "Default process group has not been initialized,"
Some information shows that I should change SyncBN to BN, is it possible in the code?
Thank you very much

deepcs233, 2023-03-14:
Hi, maybe you can refer to these two issues:
https://github.com/opendilab/InterFuser/issues/8
https://github.com/opendilab/InterFuser/issues/11

FallenMeteorite, 2023-03-15:
I tried all changes in #8, but these errors are still happened

```
bash scripts/train.sh 
/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/launch.py:188: FutureWarning: The module torch.distributed.launch is deprecated
and will be removed in future. Use torchrun.
Note that --use_env is set by default in torchrun.
If your script expects `--local_rank` argument to be set, please
change it to read from `os.environ['LOCAL_RANK']` instead. See 
https://pytorch.org/docs/stable/distributed.html#launch-utility for 
further instructions

  FutureWarning,
  File "train.py", line 1420
    if args.distributed=false:
                       ^
SyntaxError: invalid syntax
ERROR:torch.distributed.elastic.multiprocessing.api:failed (exitcode: 1) local_rank: 0 (pid: 3207920) of binary: /home/ubuntu9429/miniconda3/envs/carlause/bin/python3
Traceback (most recent call last):
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/launch.py", line 195, in <module>
    main()
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/launch.py", line 191, in main
    launch(args)
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/launch.py", line 176, in launch
    run(args)
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/run.py", line 756, in run
    )(*cmd_args)
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/launcher/api.py", line 132, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/launcher/api.py", line 248, in launch_agent
    failures=result.failures,
torch.distributed.elastic.multiprocessing.errors.ChildFailedError: 
============================================================
train.py FAILED
------------------------------------------------------------
Failures:
  <NO_OTHER_FAILURES>
------------------------------------------------------------
Root Cause (first observed failure):
[0]:
  time      : 2023-03-15_17:25:56
  host      : ubuntu9429-NH5xJNX
  rank      : 0 (local_rank: 0)
  exitcode  : 1 (pid: 3207920)
  error_file: <N/A>
  traceback : To enable traceback see: https://pytorch.org/docs/stable/elastic/errors.html
============================================================

```
Otherwise, if I use the new train.sh in #8, the parameters appears "command not found" like --train-weathers. I wonder why they are happened.

Thank you very much.

deepcs233, 2023-03-15:
Hi, for issue#8, you may need to set args.distributed = False, it means that update the  input params, instead of modifying the training code. For example, a possible solution is like adding "--distributed False" to the end of the training script or  replace torch.distributed.get_world_size() with value 1.

`Otherwise, if I use the new train.sh in https://github.com/opendilab/InterFuser/issues/8, the parameters appears "command not found" like --train-weathers. I wonder why they are happened.` . It looks like the training script is incomplete, maybe there is a blank line before `--train-weathers`.

FallenMeteorite, 2023-03-17:
Here are new errors:
```
FutureWarning,
Training with a single process on 1 GPUs.
Loading pretrained weights from url (https://github.com/rwightman/pytorch-image-models/releases/download/v0.1-weights/resnet50d_ra2-464e36ba.pth)
Model interfuser_baseline created, param count:52935567
Data processing configuration for current model + dataset:
	input_size: (3, 224, 224)
	interpolation: bicubic
	mean: (0.485, 0.456, 0.406)
	std: (0.229, 0.224, 0.225)
	crop_pct: 0.875
CNN backbone and transformer blocks using different learning rates!
165 weights in the cnn backbone, 274 weights in other modules
AMP not enabled. Training in float32.
Scheduled epochs: 35
Traceback (most recent call last):
  File "train.py", line 1844, in <module>
    main()
  File "train.py", line 1088, in main
    augment_prob=args.augment_prob
  File "/home/ubuntu9429/InterFuser/interfuser/timm/data/dataset_factory.py", line 54, in create_carla_dataset
    ds = CarlaMVDetDataset(root, towns, weathers, "det", **kwargs)
  File "/home/ubuntu9429/InterFuser/interfuser/timm/data/carla_dataset.py", line 121, in __init__
    dataset_indexs = self._load_text(os.path.join(root, 'dataset_index.txt')).split('\n')
  File "/home/ubuntu9429/InterFuser/interfuser/timm/data/base_io_dataset.py", line 18, in _load_text
    text = open(self.root_path + path, 'r').read()
FileNotFoundError: [Errno 2] No such file or directory: 'dataset/dataset_index.txt'
ERROR:torch.distributed.elastic.multiprocessing.api:failed (exitcode: 1) local_rank: 0 (pid: 24502) of binary: /home/ubuntu9429/miniconda3/envs/carlause/bin/python3
Traceback (most recent call last):
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/launch.py", line 195, in <module>
    main()
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/launch.py", line 191, in main
    launch(args)
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/launch.py", line 176, in launch
    run(args)
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/run.py", line 756, in run
    )(*cmd_args)
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/launcher/api.py", line 132, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/launcher/api.py", line 248, in launch_agent
    failures=result.failures,
torch.distributed.elastic.multiprocessing.errors.ChildFailedError: 
============================================================
train.py FAILED
------------------------------------------------------------
Failures:
  <NO_OTHER_FAILURES>
------------------------------------------------------------
Root Cause (first observed failure):
[0]:
  time      : 2023-03-17_15:58:18
  host      : ubuntu9429-NH5xJNX
  rank      : 0 (local_rank: 0)
  exitcode  : 1 (pid: 24502)
  error_file: <N/A>
  traceback : To enable traceback see: https://pytorch.org/docs/stable/elastic/errors.html
```

It seems like about dataset and agent. I learned how to create dataset_index.txt in #8, but then I found that I have a failing dataset generation before. I deleted the result in _InterFuser/results/sample_result.json_ and run run_evaluation.sh, here is a part of the errors (others are same).

```
bash leaderboard/scripts/run_evaluation.sh 
leaderboard/leaderboard/leaderboard_evaluator.py:91: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(dist.version) < LooseVersion('0.9.10'):
pygame 2.1.2 (SDL 2.0.16, Python 3.7.16)
Hello from the pygame community. https://www.pygame.org/contribute.html

========= Preparing RouteScenario_16 (repetition 0) =========
> Setting up the agent
load model: leaderboard/team_code/interfuser.pth.tar

Could not set up the required agent:
> [Errno 2] No such file or directory: 'leaderboard/team_code/interfuser.pth.tar'

Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 269, in _load_and_run_scenario
    self.agent_instance = getattr(self.module_agent, agent_class_name)(args.agent_config)
  File "/home/ubuntu9429/InterFuser/leaderboard/leaderboard/autoagents/autonomous_agent.py", line 45, in __init__
    self.setup(path_to_conf_file)
  File "leaderboard/team_code/interfuser_agent.py", line 207, in setup
    self.net.load_state_dict(torch.load(path_to_model_file)["state_dict"])
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/serialization.py", line 771, in load
    with _open_file_like(f, 'rb') as opened_file:
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/serialization.py", line 270, in _open_file_like
    return _open_file(name_or_buffer, mode)
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/serialization.py", line 251, in __init__
    super(_open_file, self).__init__(open(name, mode))
FileNotFoundError: [Errno 2] No such file or directory: 'leaderboard/team_code/interfuser.pth.tar'
> Registering the route statistics

```

I never see the file **interfuser.pth.tar** in other issues and I wonder how to get it. I noticed _auto_pilot.py_ and try to run it, and it's the most weird error today: 

> ModuleNotFountError: No module named 'team_code'

FallenMeteorite, 2023-03-17:
Here is my pip list in conda:

```
Package                  Version      Editable project location
------------------------ ------------ --------------------------------------
absl-py                  0.9.0
attrs                    19.3.0
backcall                 0.1.0
bleach                   3.1.4
cachetools               4.0.0
carla-birdeye-view       1.1.1
certifi                  2022.12.7
chardet                  3.0.4
click                    7.1.1
configparser             4.0.2
cycler                   0.10.0
decorator                4.4.2
defusedxml               0.6.0
dictor                   0.1.5
docker-pycreds           0.4.0
easydict                 1.9
elementpath              1.3.3
entrypoints              0.3
ephem                    3.7.7.1
filelock                 3.0.12
future                   0.18.2
gitdb                    4.0.2
GitPython                3.1.0
google-auth              1.35.0
google-auth-oauthlib     0.4.6
gql                      0.2.0
graphql-core             1.1
grpcio                   1.27.2
idna                     2.9
imageio                  2.8.0
imgaug                   0.4.0
importlib-metadata       1.6.0
ipykernel                5.2.0
ipython                  7.13.0
ipython-genutils         0.2.0
ipywidgets               7.5.1
jedi                     0.16.0
Jinja2                   2.11.1
joblib                   0.14.1
jsonschema               3.2.0
jupyter-client           6.1.2
jupyter-core             4.6.3
kiwisolver               1.1.0
Markdown                 3.2.1
MarkupSafe               1.1.1
matplotlib               3.0.3
mistune                  0.8.4
nbconvert                5.6.1
nbformat                 5.0.4
networkx                 2.2
notebook                 6.0.3
numpy                    1.18.3
nvidia-cublas-cu11       11.10.3.66
nvidia-cuda-nvrtc-cu11   11.7.99
nvidia-cuda-runtime-cu11 11.7.99
nvidia-cudnn-cu11        8.5.0.96
nvidia-ml-py3            7.352.0
oauthlib                 3.1.0
open3d                   0.9.0.0
opencv-python            4.2.0.34
packaging                23.0
pandas                   0.25.3
pandocfilters            1.4.2
parso                    0.6.2
pathtools                0.1.2
pexpect                  4.8.0
pickleshare              0.7.5
Pillow                   7.0.0
pip                      22.3.1
prometheus-client        0.7.1
promise                  2.3
prompt-toolkit           3.0.5
protobuf                 3.11.3
psutil                   5.7.0
ptyprocess               0.6.0
py-trees                 0.8.3
pyasn1                   0.4.8
pyasn1-modules           0.2.8
pydot                    1.4.1
pygame                   2.1.2
Pygments                 2.6.1
pyparsing                2.4.6
pyrsistent               0.16.0
python-dateutil          2.8.1
pytorch-lightning        0.7.1
pytz                     2019.3
PyWavelets               1.1.1
PyYAML                   5.3
pyzmq                    19.0.0
requests                 2.23.0
requests-oauthlib        1.3.0
rsa                      4.0
scikit-image             0.16.2
scikit-learn             0.22.2.post1
scipy                    1.4.1
Send2Trash               1.5.0
sentry-sdk               0.14.2
setuptools               67.6.0
Shapely                  1.7.0
shortuuid                1.0.1
six                      1.14.0
smmap                    3.0.1
subprocess32             3.5.4
tabulate                 0.8.7
tensorboard              2.1.1
tensorboardX             2.6
terminado                0.8.3
testpath                 0.4.4
timm                     0.4.13       /home/ubuntu9429/InterFuser/interfuser
torch                    1.13.1
torchvision              0.14.1
tornado                  6.0.4
tqdm                     4.43.0
traitlets                4.3.3
typing_extensions        4.1.1
urllib3                  1.25.8
wandb                    0.8.29
watchdog                 0.10.2
wcwidth                  0.1.9
webencodings             0.5.1
Werkzeug                 1.0.0
wheel                    0.38.4
widgetsnbextension       3.5.1
xmlschema                1.0.18
zipp                     3.1.0

```

deepcs233, 2023-03-17:
Hi!
The model can be downloaded at [here](http://43.159.60.142/s/p2CN). You could refer to Readme#Pretrain weights
 for more details.

FallenMeteorite, 2023-03-17:
> Hi! The model can be downloaded at [here](http://43.159.60.142/s/p2CN). You could refer to Readme#Pretrain weights for more details.

Thank you very much!

closed 2023-03-22 by deepcs233

FallenMeteorite, 2023-03-29:
Sorry but here are new questions
I had generated the dataset and copied them to /interfuser/dataset, then I write the dataset_index.txt in format.
Here are new errors:

```
--distributed False
/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/launch.py:188: FutureWarning: The module torch.distributed.launch is deprecated
and will be removed in future. Use torchrun.
Note that --use_env is set by default in torchrun.
If your script expects `--local_rank` argument to be set, please
change it to read from `os.environ['LOCAL_RANK']` instead. See 
https://pytorch.org/docs/stable/distributed.html#launch-utility for 
further instructions

  FutureWarning,
Training with a single process on 1 GPUs.
Loading pretrained weights from url (https://github.com/rwightman/pytorch-image-models/releases/download/v0.1-weights/resnet50d_ra2-464e36ba.pth)
Model interfuser_baseline created, param count:52935567
Data processing configuration for current model + dataset:
	input_size: (3, 224, 224)
	interpolation: bicubic
	mean: (0.485, 0.456, 0.406)
	std: (0.229, 0.224, 0.225)
	crop_pct: 0.875
CNN backbone and transformer blocks using different learning rates!
165 weights in the cnn backbone, 274 weights in other modules
AMP not enabled. Training in float32.
Scheduled epochs: 35
Sub route dir nums: 0
Sub route dir nums: 0
Traceback (most recent call last):
  File "train.py", line 1844, in <module>
    main()
  File "train.py", line 1140, in main
    pin_memory=args.pin_mem,
  File "/media/ubuntu9429/fa6442cc-89a7-4836-aed9-887481b71757/InterFuser/interfuser/timm/data/carla_loader.py", line 118, in create_carla_loader
    loader = loader_class(dataset, **loader_args)
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 344, in __init__
    sampler = RandomSampler(dataset, generator=generator)  # type: ignore[arg-type]
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/utils/data/sampler.py", line 108, in __init__
    "value, but got num_samples={}".format(self.num_samples))
ValueError: num_samples should be a positive integer value, but got num_samples=0
ERROR:torch.distributed.elastic.multiprocessing.api:failed (exitcode: 1) local_rank: 0 (pid: 11901) of binary: /home/ubuntu9429/miniconda3/envs/carlause/bin/python3
Traceback (most recent call last):
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/launch.py", line 195, in <module>
    main()
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/launch.py", line 191, in main
    launch(args)
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/launch.py", line 176, in launch
    run(args)
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/run.py", line 756, in run
    )(*cmd_args)
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/launcher/api.py", line 132, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
  File "/home/ubuntu9429/miniconda3/envs/carlause/lib/python3.7/site-packages/torch/distributed/launcher/api.py", line 248, in launch_agent
    failures=result.failures,
torch.distributed.elastic.multiprocessing.errors.ChildFailedError: 
============================================================
train.py FAILED
------------------------------------------------------------
Failures:
  <NO_OTHER_FAILURES>
------------------------------------------------------------
Root Cause (first observed failure):
[0]:
  time      : 2023-03-29_18:36:38
  host      : ubuntu9429-NH5xJNX
  rank      : 0 (local_rank: 0)
  exitcode  : 1 (pid: 11901)
  error_file: <N/A>
  traceback : To enable traceback see: https://pytorch.org/docs/stable/elastic/errors.html
============================================================
```

I think the key of the error is ` num_samples=0 `.

Otherwise, is it normal that the result of model is failed (such as unfinished route) when generating the dataset?

deepcs233, 2023-03-29:
From your log:  `Sub route dir nums: 0`, it means that the trainer can not find the dataset, so it crashed.
Maybe this issue(https://github.com/opendilab/InterFuser/issues/20#issuecomment-1441362950) can help you.

FallenMeteorite, 2023-04-03:
I noticed that the subfolder of weather-0/data/routes_town05_long_03_18_00_14_08 is "meta", and pictures in the folder are same as Demo Video in readme, not "rgb_front" or so on in #18 and #8. 
What should I do if I want to generate data like them?

deepcs233, 2023-04-03:
It's because you ran the evaluation script, not data collection script. Please refer to https://github.com/opendilab/InterFuser#data-generation-with-multiple-carla-servers for multiple carla servers or https://github.com/opendilab/InterFuser#data-generation-with-a-single-carla-server for a single carla server