opendilab/InterFuser #28: Question about position of ego vehicle relative to waypoints

Jmacal, 2023-04-12:
Hi, I am experimenting with different lateral and longitudinal controllers and some require the x and y position of the ego vehicle. I know that in the object density map the position of the ego vehicle is (0,0). Are these waypoints of the same coordinate system as the object density map, meaning the x,y of the ego vehicle is (0,0) relative to these waypoints? Or am I mistaken?

Thanks for any help you can provide

deepcs233, 2023-04-12:
Hi!
The coordinate system of ego vehicle is the same as the system of the object density map. （0,0) of the ego vehicle is mapped to (10, 2) in the coordinate system of object density map. (0, 0) here means the center point of the ego vehicle, and the size of the object density map is (20, 20). 

You may refer the right part of the picture in the main page of this repo, and will find that the bottom point in the middle is the rear of the ego car, not the center position. With this design, we hope the object density map could cover the objects 
near the rear of the car to improve safety. 

About the detail of code, may refer to https://github.com/opendilab/InterFuser/blob/e5a6a914d0699611601b9429e71391a91fbba798/interfuser/timm/data/det_utils.py

deepcs233, 2023-04-13:
1）There is no direct relationship between these two systems.
2)  This is right: the way points have their own coordinate system with the ego vehicle at (0,0). and a waypoint in front the vehicle in this case would be (0,5).
3) To utilize the object density map for avoiding collision, we need to map the predicted waypoints to the coordinate of the density map.