opendilab/InterFuser #31: pygame.error: No available video device

SHITIANYU-hue, 2023-04-30:
Hello 

May i know why it will have this issue, thanks!

  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 269, in _load_and_run_scenario
    self.agent_instance = getattr(self.module_agent, agent_class_name)(args.agent_config)
  File "/media/changquan/data0/sty/InterFuser/leaderboard/leaderboard/autoagents/autonomous_agent.py", line 45, in __init__
    self.setup(path_to_conf_file)
  File "leaderboard/team_code/interfuser_agent.py", line 162, in setup
    self._hic = DisplayInterface()
  File "leaderboard/team_code/interfuser_agent.py", line 50, in __init__
    (self._width, self._height), pygame.HWSURFACE | pygame.DOUBLEBUF
pygame.error: No available video device
> Registering the route statistics

========= Preparing RouteScenario_25 (repetition 0) =========
> Setting up the agent

Could not set up the required agent:
> No available video device

Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 269, in _load_and_run_scenario
    self.agent_instance = getattr(self.module_agent, agent_class_name)(args.agent_config)
  File "/media/changquan/data0/sty/InterFuser/leaderboard/leaderboard/autoagents/autonomous_agent.py", line 45, in __init__
    self.setup(path_to_conf_file)
  File "leaderboard/team_code/interfuser_agent.py", line 162, in setup
    self._hic = DisplayInterface()
  File "leaderboard/team_code/interfuser_agent.py", line 50, in __init__
    (self._width, self._height), pygame.HWSURFACE | pygame.DOUBLEBUF
pygame.error: No available video device
> Registering the route statistics
> Registering the global statistics

deepcs233, 2023-04-30:
Hi！
The issue may be caused by you run the code on a Linux server that doesn't have a visuable window.
The following link may help you: https://stackoverflow.com/questions/15933493/pygame-error-no-available-video-devicehttps://stackoverflow.com/questions/15933493/pygame-error-no-available-video-device
Or you can just comment the code related to `self._hic`.

SHITIANYU-hue, 2023-05-01:
Hi, thanks for your reply, yes, i tried to find a computer with visuable window, I run this in one terminal
`carla/CarlaUE4.sh --world-port=2000 -opengl`

then i run this in another terminal : 
`./leaderboard/scripts/run_evaluation.sh`

then, it says carlaUE4 is not responding, may i know if it is possible to accelerate the rendering process? 

[image]

However，after a while, around 10 mins, it could run normally: 
[image]

deepcs233, 2023-05-07:
Sorry, we have on way for this problem. Maybe you could change a better GPU card for this.