opendilab/InterFuser #33: Some questions about the tables in the paper

tulerfeng, 2023-05-06:
Hi, thanks for your wonderful work. I have got one small quetions.

I find that, Table 2, Table 3, Table 4 are all the evaluation on Town05 benchmark. But why in Table 2 and Table 3, InterFuser derive 51.6 driving score on Town 05 Long, while in Table 4, , InterFuser derive 68.31 driving score ?

I'm looking forward for your reply !

deepcs233, 2023-05-07:
Hi, one is for sota, the one is for ablation study. They got the different results because they don't share the same baseline.

tulerfeng, 2023-05-07:
Thanks for your answer. Would you please tell me what's the difference between the two baselines ? Is the training data are different or the training epoch are different, or others ?

deepcs233, 2023-05-07:
For the convenience of the ablation study, we mainly used a smaller training dataset and a smaller model size.

tulerfeng, 2023-05-08:
Thanks for your reply, my question has been addressed.

closed 2023-05-08 by tulerfeng