opendilab/InterFuser #34: Town05 Benchmark

cstlove, 2023-05-24:
In the file you provided, I found the long route of 05town (routes_town05_long.xml), but there is no short route in the same folder. I found routes_town05_short.xml in the validation_routes folder, but it is mentioned in your paper The short route file consists of 10 routes, while routes_town05_short.xml consists of 32 routes.

deepcs233, 2023-05-24:
Sorry for that, 10 routes here denotes the route number of the long route. We will fix it in our updated paper in Arxiv.

cstlove, 2023-05-24:
> Sorry for that, 10 routes here denotes the route number of the long route. We will fix it in our updated paper in Arxiv.

I am reproducing your paper code recently, so the short routes data set you tested in the town05 benchmark is composed of 32 routes (routes_town05_short.xml)?