opendilab/InterFuser #35: question about model

EcustBoy, 2023-05-29:
Hi~ author,  I haven't found the detailed network structure python script in your code, I wanna ask have you open source this file? I'll thanks for your reply~

deepcs233, 2023-05-29:
Hi!
(1) It can be found in https://github.com/opendilab/InterFuser/blob/main/interfuser/timm/models/interfuser.py
(2) No, the traffic light predictor can utilize the feature from all sensors in our framework.
(3) The traffic light detection has a very big impact on the final result. It will multiply 0.6 into the final score if the agent run the red light. Our framework is not based on imitation learning, so we haven't done similiar ablation experiments.

EcustBoy, 2023-05-29:
> 

thanks for your reply ~  about question (3),  the "imitation learning" in my word means the model need expert traj as label~ and I think interfuser is done like this(the additional safety controller can be seen as a post-process step I think)

And I just found the "is_junction" flag is also learned in your model,  I think "is_junction" flag and traffic light state info must be used together, right?

Beside, imaging such a scenario case (although with low probability)：ego car is runing close to junction and the traffic light is green, but the model outputs the wrong behavior to slow down and brake,  can interfuser model give ego car the throttle to continue accelarating? (I seem to only find the slow down behavior given by safety controller for collision avoidance in the code, but I'm not sure about such scenario case)

deepcs233, 2023-05-31:
Hi
(1) Interfuser doesn't imitate the expert traj, it only predictes the driving route.
(2) Yes, if "Is_junction" is True, we will ignore the output of the traffic light predictor.
(3) According to response(1), our method only output the driving route (may be dangerous). So if the light is green there are no other dangerous situations,the ego car will not slow down.

EcustBoy, 2023-05-31:
> Hi (1) Interfuser doesn't imitate the expert traj, it only predictes the driving route. (2) Yes, if "Is_junction" is True, we will ignore the output of the traffic light predictor. (3) According to [...]

Thanks for your reply~
About answer (2), according to code, I think the output of the traffic light predictor is considered only when "Is_junction" is True, right?

[image]