opendilab/InterFuser #42: Script for collecting data

a1wj1, 2023-06-30:
[image]
When we collect data, we run the leaderboard_ evaluator.py, this process will also result in the evaluation of the data, right? Because when I generated the data, I obtained the results shown in the figure. If I only performed data generation without evaluation, what would I do?

deepcs233, 2023-06-30:
Hi!

> When we collect data, we run the leaderboard_ evaluator.py, this process will also result in the evaluation of the data, right?
Yes

> If I only performed data generation without evaluation, what would I do?
There is no need to do so, because data generation and evaluation are one same process.  We just use the framework of the evaluation and collect data in it at the same time. It does not incur additional resource/time consumption.

a1wj1, 2023-06-30:
Thank you very much!

SExpert12, 2024-05-24:
@a1wj1 ,
Hi, I am not able to collect the data set. Can you help how to do that?
I have also run this file but I am stuck at 

from scenario_runner.srunner.scenariomanager.carla_data_provider import *
ModuleNotFoundError: No module named 'scenario_runner'

To correct this, I modified code like this.
import sys
import os

sys.path.append("InterFuser-main/scenario_runner/srunner/scenariomanager")
from .scenariomanager.carla_data_provider import *

, but it is not working yet.

Thanks