opendilab/InterFuser #44: Waypoint loss function

a1wj1, 2023-06-30:
In the Waypoint loss function, the waypoints generated by the expert agent are the waypoints used during data collection. Is that right？

a1wj1, 2023-06-30:
Also, I am not very familiar with the concept of moving average for trajectory prediction mentioned in the paper. Could you please explain it？Thank you!

deepcs233, 2023-07-03:
Hi!

> In the Waypoint loss function, the waypoints generated by the expert agent are the waypoints used during data collection. Is that right？

No, the ground truth waypoints when computing waypoint loss are generated by the global planner during data collection. The gt waypoints are not the actual waypoints that the ego-car used. They can be viewed as "planned route", not the actual "driving route". Our model only predicts "planned route", and use other outputs (Object density map, Traffic light prediction. etc.) to refine this route to actual "driving route".

> Also, I am not very familiar with the concept of moving average for trajectory prediction mentioned in the paper. Could you please explain it？Thank you!

We use a tracker to track other agents' historical statusto get a more accurate motion prediction. Here we use "moving average" mechanism to update their status (like velocity, heading, etc.) when we get a last prediction of their status. We use this because the prediction is not very accurate as it is based on single-frame input. We need to  utilize "temporal consistency" to refine the prediction.