opendilab/InterFuser #45: weather used

a1wj1, 2023-07-01:
In the paper, I did not see how many types of weather were used in the dataset. Is it the same as the code, using 14 types of weather?

a1wj1, 2023-07-01:
I have a few questions after reading the paper. Can you help me answer them? Thank you：

1. Are the variables shown in the figure also manually set parameters?
[image]

2. In the Limitation, you mentioned:
Extending the current deterministic form to a probabilistic form can also help to deal with the uncertainty and multi-modality in detection and prediction.
But I don't quite understand where the deterministic form in the method is reflected?

deepcs233, 2023-07-03:
Hi,
about the weather , please refer to https://github.com/opendilab/InterFuser/issues/41#issuecomment-1614134553

> Are the variables shown in the figure also manually set parameters?

Yes

> But I don't quite understand where the deterministic form in the method is reflected?

InterFuser predicts other agents' status, future waypoints with a deterministic form. A deterministic form here means a discrete value, not a distribution (like gauss, laplace, etc. ).