opendilab/InterFuser #47: Training

a1wj1, 2023-07-04:
Hi，when training，I see parameters:
--dataset carla --train-towns 1 2 3 4 6 7 10  --val-towns 5  --train-weathers 0 1 2 3 4 5 6 7 8 9  --val-weathers 10 11 12 13 
But For each town, it is divided into long, short, and tiny. Do all three need to collect data and use it? And why is the weather for training not 21, but the top 14?

deepcs233, 2023-07-04:
> Do all three need to collect data and use it?

Yes

>  And why is the weather for training not 21, but the top 14?
We  recommend to use all 21 weather.

a1wj1, 2023-07-05:
Is the effect of your paper using 21 different weather conditions？

deepcs233, 2023-07-05:
Yes

a1wj1, 2023-07-05:
Is there a download path for the collected dataset? Because we collect data very slowly on the experimental platform. Thank you!