opendilab/InterFuser #48: collect data from the CARLA leaderboard benchmark

a1wj1, 2023-07-04:
Thank you. May I ask how to modify the content in leaderboard/scripts/run_evaluation.sh if I want to collect data from the CARLA leaderboard benchmark?

[image]

Is it correct to only modify SCENARIOS and ROUTES？as shown in the figure.

deepcs233, 2023-07-04:
Yes, you are right!

deepcs233, 2023-07-05:
The more the better, `routes_training.xml` has the most proportion of the data.

a1wj1, 2023-07-05:
According to the indicators in your paper, did you run three xmls together？

deepcs233, 2023-07-26:
No, we collect the drivinf data based on other xmls (please refer to repo#data collection)