opendilab/InterFuser #50: All in one docker

Mariusmarten, 2023-07-13:
Hi, thank you for making the code available!

I would like to run the trained InterFuser model on a small set of pre-collected CARLA data. Can I download such data from somewhere without collecting them myself? How would you go about setting up such a lightweight setting?

Mariusmarten, 2023-07-24:
Thank you for the information. Do I understand correctly that with those weights: http://43.159.60.142/s/p2CN I will not need to train the model at all?

deepcs233, 2023-07-25:
Hi!
If you just want to visualize/test the driving capabilities on your data for convenience, it's ok to use this pretrain weights. But it you want to eval the performance in some benchmarks, you need to re-collect the correspondind dataset and re-train it.

Mariusmarten, 2023-08-13:
Hi! Thank you for the response. How good are the pre-trained weights? Are they close to the actual model performance?