opendilab/InterFuser #54: Werid behavior before stop sign

JkhwungII, 2023-07-24:
Hi, I'd like to know is it normal for ego car to inch forward then stop like this when stop signs are detected. 

https://github.com/opendilab/InterFuser/assets/53292322/ec4479a2-2ce9-44b0-86cf-911b4b428932

And if its not normal how can i fix it. I trained the model with pre-train weight offered loaded. Batch size 2 and 1 epoch with 600,000 frames.

deepcs233, 2023-07-25:
Hi!
The problem does not arise from the model. The `werid` behavior is defined in our controller module https://github.com/opendilab/InterFuser/blob/dcc2c20bba2c8656773baab5a552a27fb74e86dc/leaderboard/team_code/interfuser_controller.py#L140

Because it's very strict to determine whether or not to obey a stop sign in carla scenarios and it's hard for the agent to stop at the correct area. We design and develop this `werid` behavior to let the ego-car stop serveral times to get a better performance about stop sign score.

For more details, you can refer to https://github.com/opendilab/InterFuser/issues/25#issuecomment-1481035474

JkhwungII, 2023-07-25:
Many thanks. However is it normal for the model to detect stop sign that far away. The stop sign prob was at 1.0 when the car first stopped. I'd like to know this because the model with pretrain weight would stop way ahead of the actual junction and the problem was solved with futher training. I wonder is this the same with stop sign.

deepcs233, 2023-07-26:
Hi!
1. Because it's very hard for an end-to-end model to recognize the distance to the stop sign (besides, the actual trigger area defined in Carla simulator is usually different from the location of stop sign). We augment the stop sign data to make it more sensetive to detect the stop sign.
2. Concerning the issue of the vehicle stopping way ahead of the actual junction, it's likely due to the ego-agent detecting the red light from a significant distance, or it might erroneously recognize a green light as red.

JkhwungII, 2023-07-27:
Many thanks.