opendilab/InterFuser #58: map issue

a1wj1, 2023-08-10:
Hello, how do you consider the situation where the current frame of an object exists, but the next frame does not exist in the map?

deepcs233, 2023-08-12:
Hi!
Our framework also predicts the velocity and the heading when detect the other objects. With these information, we can estimate their future positions. If they are far away from the ego-car, we will not consider them again.

a1wj1, 2023-08-13:
Hello, what I mean is during the training phase, how to make predictions when training data if an object in the future frame no longer exists

deepcs233, 2023-08-13:
In our current pipeline, the framework receives single-frame input and predict single-frame output. Or you can refer to ReasonNet to modify the framework to let it predict multiple frames in the future.