opendilab/InterFuser #65: Not able to generate data

atanuc13, 2023-09-01:
I was trying to generate datasets from 1 gpu.

 So, initially I started the carla server on 1 gpu.
[image]

Then I was running data_collection/bashs/weather-0/routes_town01_tiny.sh file but unfortunately getting segmentation fault error. 
[image]

Can anyone please help, if you have also faced similar issue while running on single gpu.

deepcs233, 2023-09-01:
Hi!
Maybe you can try to restart carla server and re-run the data_collection script. Also check the ports of these two files are the same.

naren200, 2023-10-31:
Hello!

I have encountered the same issue! Searched this issue in Carla Github Issues page but no luck!

## Steps followed to reproduce
### Initiating Carla Simulator
[image]

### Running the run_evaluation script
[image]
[image]

### Issue: Segmentation Fault;
[image]

```
LowLevelFatalError [File:Unknown] [Line: 3762] 
Failed to link program [Program V_3AA5F7CCAD2B351344BE53DDA22E8BAEBA78720A_P_ABADFE631118A2CF0719C49D3F936DABF0CE8865]. Current total programs: 238, precompile: 0
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=149952 LargeMemoryPoolOffset=281088 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

```
Human Agent is blank
[image]

Any suggestions would be more than welcome, thanks. 

Regards,
Naren