opendilab/InterFuser #67: Ego vehicle’s (goal) location

Mariusmarten, 2023-09-06:
How is the ego vehicle’s goal location stored in the data? And does the vehicle obtain ground truth coordinate information, or is the location estimated?

I found the following values for the vehicle position:
`"gps_x": -330.58303211256947,
    "gps_y": 62.00891884672404,
    "x": 62.00891876220703,
    "y": 330.5678405761719`
stored in the `weather-X/data/routestown01../measurements/0000.json` file.

deepcs233, 2023-09-06:
Hi!
Your understanding is correct. The "gps_x" and "gps_y" here represent the location given by CARLA Leaderboard. The "x" and "y" here denote the location provided from CARLA simulator.

Mariusmarten, 2023-09-06:
Thank you for the quick response. So, to be clear "gps_x" and "gps_y" are the goal locations, and "x" and "y" define the current ground truth position of the vehicle?
And during one sequence the ground truth position is only provided once (at the beginning of the sequence) and then internally updated using the model's actions?

deepcs233, 2023-09-06:
> Thank you for the quick response. So, to be clear "gps_x" and "gps_y" are the goal locations, and "x" and "y" define the current ground truth position of the vehicle?

No, they are all the current ground truth position of the vehicle, but in different coordinates. They can be converted to each other. "x" = "gps_y", "y" = -"gps_x"

> And during one sequence the ground truth position is only provided once (at the beginning of the sequence) and then internally updated using the model's actions?

No, these measurements are provided in the whole period. You can check the 0001.json, 0002.json ...

Mariusmarten, 2023-09-11:
Thanks a lot, I understand now!

One final question:  I dont see speed or commands as model outputs. Where / How are the resulting output waypoints translated back into actions for the CARLA environment? Or is the model only evaluated in an offline fashion?

deepcs233, 2023-09-11:
Hi!
About this question. i recommend you to refer this page: https://github.com/opendilab/InterFuser/blob/main/leaderboard/team_code/interfuser_agent.py

In a word, the agent controller receives the path and traffic prediction of the model. And then compute the safe route that the ego-car can drive. Finally, the code will convert the driving route into the actual actions.

Mariusmarten, 2023-09-13:
Very helpful! I try to run the model on offline data, but I need help finding where the InterfuserAgent class (from the interfuser_agent.py) is instantiated and obtaining its input from the InterFuser decoder. Can you point me to the relevant file?