opendilab/InterFuser #69: It cannot work

caixxuan, 2023-09-26:
If i input the following command:

(interfuser) cai@cai:~/InterFuser$ bash ./leaderboard/scripts/run_evaluation.sh

then output the result and get out and no other things happen!
pygame 2.1.2 (SDL 2.0.16, Python 3.7.16)
Hello from the pygame community. https://www.pygame.org/contribute.html
> Registering the global statistics

caixxuan, 2023-09-26:
This is the first running:
...
========= Preparing RouteScenario_41 (repetition 0) =========
> Setting up the agent

Could not set up the required agent:
> Unknown model (interfuser_baseline)

Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 269, in _load_and_run_scenario
    self.agent_instance = getattr(self.module_agent, agent_class_name)(args.agent_config)
  File "/home/cai/InterFuser/leaderboard/leaderboard/autoagents/autonomous_agent.py", line 45, in __init__
    self.setup(path_to_conf_file)
  File "leaderboard/team_code/interfuser_agent.py", line 204, in setup
    self.net = create_model(self.config.model)
  File "/home/cai/anaconda3/envs/interfuser/lib/python3.7/site-packages/timm/models/_factory.py", line 110, in create_model
    raise RuntimeError('Unknown model (%s)' % model_name)
RuntimeError: Unknown model (interfuser_baseline)
> Registering the route statistics
> Registering the global statistics

caixxuan, 2023-09-26:
[image]

caixxuan, 2023-09-27:
More question:How to modify the scenario I want?
e.g. I want to test the interfuser_agent in the intersection scenario by setting some NPC vehicles controlled by RL, the NPC's initial position and velocity can be defined anyway

deepcs233, 2023-09-27:
Hi!

> e.g. I want to test the interfuser_agent in the intersection scenario by setting some NPC vehicles controlled by RL, the NPC's initial position and velocity can be defined anyway

If the scenario that you want to test includes some agents controlled by RL, i think it's a very difficult thing. You may need to rewrite the whole benchmark framewoke.

If the scenario that you want to test includes some agents controlled by the special rule, you may need to create a new scenario file in the existing framework. You can refer to my other repo: DOS, i have created four types of scenario in that repo.

caixxuan, 2023-09-28:
> Hi!
>
>
> If the scenario that you want to test includes some agents controlled by RL, i think it's a very difficult thing. You may need to rewrite the whole benchmark framewoke.
>
> If the scenario that [...]

I am interesting to create the waypoints file (.xml) to guide the agent to follow, how do i make one route waypoints I want?

deepcs233, 2023-09-28:
Hi！
Here is the website to view the xodr map file:
https://odrviewer.io/

You can use it to load carla map files and choose the waypoints sequence that you want the car to follow.

deepcs233, 2023-09-28:
But i remember the coordinate system of the carla map and the website is differnet, maybr (x, y) -> (y, -x)? You can try it.

caixxuan, 2023-09-29:
> Hi！ Here is the website to view the xodr map file: https://odrviewer.io/
> 
> You can use it to load carla map files and choose the waypoints sequence that you want the car to follow.

this website seems to output .obj file only , I don't know how to use it. Your .xml files in ./leaderboard/data are made by this website?

deepcs233, 2023-09-29:
We do not use the outpur of the website directly. We upload the carla map file (*.xodr, which can be dumpped from the carla python api) and we can get the visualization of carla map. Finally, we choose the positions as the start point and end point (maybe have some other passing waypoints), and write them into the xml files.

caixxuan, 2023-10-09:
> We do not use the outpur of the website directly. We upload the carla map file (*.xodr, which can be dumpped from the carla python api) and we can get the visualization of carla map. Finally, we [...]

thanks! I have met another problem: the interfuser agent vehicle is set to cruise from A point to B point, the first time it can work properly, but when I use the CARLA 'set_transform' function to move the agent from the its location when it's on its way, it don't work properly. e.g. If the first time it choose to turn left at a intersection, after 'set_transform' to the initial location before the agent complete its route, the second time the agent may will not turn left and don't work normally. I am wondering the route completion remark or other things is not reset but I dont know how to adjust. So how to solve this problem

deepcs233, 2023-10-10:
Hi!
The function 'set_transform' is always used in the position initialization of the backgroung agents (like the walkers, running-light cars). The agent (ego-car) will cruise the route which is generated by the A* algorithm when we start the route. Then if you change the agent's poisition, it can not follow the previous generated navigation route (which may have been passed part o it ) . If you want the agent follow the accurate route, you can design more dense navigation points in the xml files. You can also refer the code related to the scenario, like https://github.com/opendilab/DOS/blob/main/scenario_runner/srunner/scenarios/cut_in_active_object.py