opendilab/InterFuser #72: [Clarification] Data Generation Question

naren200, 2023-10-08:
---
name: Data Generation Question
about: Seek clarification on the data generation process
title: "[Clarification] Data Generation Question"
labels: question
assignees: 
---

## Description & Step-to-reproduce

**Describe the Issue:**
Please describe your question or the issue you're facing regarding the data generation process. Be as specific as possible, and include any relevant code or documentation snippets.

For example:
In the documentation, I found instructions for data generation in two sections:

1) Under "Run batch-run scripts of the town and route type that you need to collect," there is a command mentioned:

```
bash data_collection/batch_run/run_route_routes_town01_long.sh

```
2) In the section "Data Generation with a single CARLA Server," the following commands are mentioned:
```
carla/CarlaUE4.sh --world-port=2000 -opengl
./leaderboard/scripts/run_evaluation.sh
```

## Context

**Repository Link:**
[Documentation where I found these instructions](https://github.com/opendilab/InterFuser).

**Additional Information:**
Include any additional context or information that might help someone understand your question better.

## Screenshots & Expected Behavior
- Last Few Outputs of Long for all weather data generation through (1)st method. It took around 6 hours for 1 long route, giving a feel of generated data.
[image]

- After running the (2) Data generation option. Is the data generation completed? Took less than a minute to generate, is that all?
[image]If applicable, add screenshots or images to illustrate the issue.

## Environment

- Operating System: Linux Ubuntu 20.04
- Python version : 3.7 as downloaded through Carla.
- Docker: Running all the files through my system but not Docker at all

deepcs233, 2023-10-10:
Hi!

> After running the (2) Data generation option. Is the data generation completed? Took less than a minute to generate, is that all?

You need to delete the results/sample_result.json. Because the generation code will resume the last collection, or you can set the flag of RESUME as False.

naren200, 2023-10-14:
Hey! Thank you for the quick reply. 
I have tried to remove the file and re-run the command. The output's the same and the time of execution is less than 10 seconds. I don't think it ran.
Kindly, guide me through this issue. If you need more clarification or files, please feel free to ask.

[image]

Thanks

deepcs233, 2023-10-14:
Hi!

You may check this file: https://github.com/opendilab/InterFuser/blob/main/data_collection/base_script.sh#L17
If it is correct, you can check the term `TEAM_AGENT` in the sub-bash file o data_collection/batch_run/run_route_routes_town01_long.sh.

Here we only use the auto-poilt agent to collect the data, not the interfuser agent. So the interfuser.pth.tar is not needed when we collect the data.