opendilab/InterFuser #74: Request for Ideas to Enhance an Existing Paper in Open Source

naren200, 2023-10-15:
## Request for Ideas

Hello open-source community & DeepCS233,

I hope this message finds you well. I am reaching out to seek your input and ideas regarding an exciting project I am planning. 

My intention is to use this paper as a baseline and work on top of it to contribute valuable enhancements and additions. However, I would greatly appreciate your insights, ideas, and suggestions on what specific aspects or directions I could explore to make this project truly beneficial for the open-source community.

### What I'm Seeking

I am interested in any ideas, recommendations, or potential features that you believe would complement the existing paper and contribute to the open-source community. These ideas could include:

- **Improvements:** Suggestions for improving existing methods or algorithms.
- **Extensions:** Ideas for extending the paper's concepts into new areas.
- **Other Ideas:** Any other creative thoughts you may have to enhance the project.

Your insights, whether big or small, can help shape the direction of this initiative and contribute to the open-source community's growth.

### How to Contribute
If you have any ideas or suggestions, please share them by commenting on this issue. I am open to engaging in discussions, collaborating, or even creating sub-projects if there are specific aspects that require dedicated attention.

Thank you for taking the time to read this request. Your contributions will be highly valued, and I look forward to working together to expand and improve upon the foundation provided by the paper.

Best regards,
Naren

deepcs233, 2023-10-20:
Hello Naren,

Thank you for reaching out. For the project you mentioned, I suggest exploring the temporal direction, specifically tackling long-sequence decision-making problems such as overtaking. This could offer valuable enhancements and extensions to your initiative.

Looking forward to potential collaboration and witnessing the progress of your project.

Best regards,
Hao

naren200, 2024-02-15:
Hi [Dr. Hao](https://github.com/deepcs233),

I am currently continuing your project, and I am exploring aspects of the temporal direction. Here, I am finding it difficult to show the results of my work, like Driving Score, Route Completion, and Infraction penalty. As, I have to showcase results at the end of my project. 

I ran the model several times and got several checkpoints under "/interfuser/output/*", and the last checkpoint seems to drive the car well, but I am unsure how to calculate the parameters. I know about the instructions given here, [https://leaderboard.carla.org/](https://leaderboard.carla.org/). But I am here to ask how to let the model spit out the results automatically. 

Another less priority help, is there any documentation you suggest or which files to change under this repository to accomplish this temporal logic performing long-term decisions? I am new to AI or research; you can view my [profile](https://github.com/naren200). I will make use of any help/ documentation you throw at me! I am open to connecting through social media and building a network.

Thanks,
Naren

Portfolio: https://naren200.github.io

deepcs233, 2024-02-17:
Hi!
After finishing the model training, you still need to evaluate it in a closed-loop manner to get the final results, such as the `driving score`, and `route completion`. For your reference: https://github.com/opendilab/InterFuser?tab=readme-ov-file#evaluation

For the second question, I have some suggestions:

- model definition: https://github.com/opendilab/InterFuser/blob/main/interfuser/timm/models/interfuser.py
- training dataset: https://github.com/opendilab/InterFuser/blob/main/interfuser/timm/data/carla_dataset.py
- agent definition: https://github.com/opendilab/InterFuser/blob/main/leaderboard/team_code/interfuser_agent.py