opendilab/InterFuser #78: Setup error

abol-karimi, 2023-11-14:
I followed the setup instructions, but I get the following errors:
```
ERROR: Ignored the following versions that require a different python version: 1.25.0 Requires-Python >=3.9; 1.25.0rc1 Requires-Python >=3.9; 1.25.1 Requires-Python >=3.9; 1.25.2 Requires-Python >=3.9; 1.26.0 Requires-Python <3.13,>=3.9; 1.26.0b1 Requires-Python <3.13,>=3.9; 1.26.0rc1 Requires-Python <3.13,>=3.9; 1.26.1 Requires-Python <3.13,>=3.9; 1.26.2 Requires-Python >=3.9; 3.2 Requires-Python >=3.9; 3.2.1 Requires-Python >=3.9; 3.2rc0 Requires-Python >=3.9; 3.8.0 Requires-Python >=3.9; 3.8.0rc1 Requires-Python >=3.9; 3.8.1 Requires-Python >=3.9; 8.13.1 Requires-Python >=3.9; 8.13.2 Requires-Python >=3.9; 8.14.0 Requires-Python >=3.9; 8.15.0 Requires-Python >=3.9; 8.16.0 Requires-Python >=3.9; 8.16.1 Requires-Python >=3.9; 8.17.0 Requires-Python >=3.9; 8.17.1 Requires-Python >=3.9; 8.17.2 Requires-Python >=3.9
ERROR: Could not find a version that satisfies the requirement open3d==0.9.0.0 (from versions: 0.10.0.0, 0.11.0, 0.11.1, 0.11.2, 0.12.0, 0.13.0, 0.14.1, 0.15.2, 0.16.0, 0.17.0)
ERROR: No matching distribution found for open3d==0.9.0.0
```

deepcs233, 2023-11-14:
Hi!
Maybe you can delete this package in the requirement.txt?
If the code cannot run, then you can pip install it again.

deepcs233, 2023-11-15:
maybe you can try:

pip install -U pip setuptools