opendilab/InterFuser #80: Evaluation performance for Town05 Long from the paper table

JunyongYun-SPA, 2023-11-27:
[image]

The above table is the performance table for town05 Long in the paper supplement. However, when evaluated with the pretrain weight provided by GitHub, the corresponding performance is not achieved. (driving score: 57.447 / route completion: 89.142)

How can I get that performance?

deepcs233, 2023-11-27:
Hi!
The released model was only trained on a subset of the training dataset. You could collect more dataset from other towns and train it. Some hyper-parameters in the controllers may be adjusted according to the autual performance.

JunyongYun-SPA, 2023-11-28:
Thank you for your quick response!

YilinGao-SHU, 2023-12-22:
Thank you very much for your reply, I have another question here, why the accuracy in the ablation experiments seems to be less 
 than the one in the appendix when the same comparison is done on the Town05 Long, is there some detail in the paper that I am missing here?
[image]

deepcs233, 2023-12-22:
Hi!

Our experiments are conduct on LangAuto-Long, instead of Town05-long. May
be some misunderstanding here :)

YilinGao-SHU, 2023-12-22:
> […](#)

Hi！
This table comes from page 7 of "Safety-Enhanced Autonomous Driving Using Interpretable Sensor Fusion Transformer". Should there be no LangAuto yet?

deepcs233, 2023-12-22:
Hi!
Sorry for the wrong reply, i just used my phone to reply you. And I thought I replied to you in that LMDrive issue. The table in pgae7 didn't share the same setting in the appexdix. Considering we need to conduct many ablation experiments, we only used a subset of our training set for the ablation studies, which may cause a performance drop.

YilinGao-SHU, 2023-12-22:
> Hi! Sorry for the wrong reply, i just used my phone to reply you. And I thought I replied to you in that LMDrive issue. The table in pgae7 didn't share the same setting in the appexdix. Considering [...]

Thanks so much for you reply!