opendilab/InterFuser #82: Evaluation

No4x, 2023-12-30:
Hello, I used 1M frames to train model and the result of my evaluation is "score_composed": 42.566437375424,"score_penalty": 0.45978602031298343,"score_route": 89.19403995945298. The route completion is already good but the infraction  score is poor. Is this acceptable? Or is there any way to improve performance?

deepcs233, 2024-01-03:
Hi! 
Can you provide which benchmark did you use to get these results? Or could you provide more details?

No4x, 2024-01-03:
> Hi! Can you provide which benchmark did you use to get these results? Or could you provide more details?

I used town05 long as benchmark.

I have another question about tools to remove hazard frames. Is "blocked_stat.txt" from batch_stat_blocked_data.py equivalent to "haze_stat.txt"?  I directly use it as "haze_stat.txt" but encountered some issues.

deepcs233, 2024-02-17:
Hi!
The two files are equivalent. Sorry for the confusion, I haven't cleaned the related code about the data tools.

closed 2024-02-18 by No4x