opendilab/InterFuser #83: Questions about running Carla on GPU servers

YilinGao-SHU, 2024-01-04:
Thanks so much for your work! But I got some puzzle problems when I try to run Carla on a GPU Server with 8 GPUs, when I use the following commands to start a Carla server, all the process run on gpu 0 like this. 
`CUDA_VISIBLE_DEVICES=0 ./CarlaUE4.sh --world-port=20000 -carla-server &
CUDA_VISIBLE_DEVICES=1 ./CarlaUE4.sh --world-port=20002 -carla-server &`
Have u ever met this mistake?
[image]

deepcs233, 2024-01-04:
Hi!
Could you show the output of `nvidia-smi`? I can not understand the screenshot :)

YilinGao-SHU, 2024-01-05:
> Hi! Could you show the output of `nvidia-smi`? I can not understand the screenshot :)

Of course! The usage of the GPUs looks like this. It can be found that all Carla process run on GPU 0.
[image]
Looking forward to your reply~

deepcs233, 2024-01-05:
Hi, this is my `nvidia-smi` output and it looks ok:
[image]

the corresponginf script:
```bash
CUDA_VISIBLE_DEVICES=0 ./CarlaUE4.sh --world-port=2000 -opengl  &
sleep 1
CUDA_VISIBLE_DEVICES=1 ./CarlaUE4.sh --world-port=2002 -opengl  &
sleep 1
CUDA_VISIBLE_DEVICES=2 ./CarlaUE4.sh --world-port=2004 -opengl  &
sleep 1
CUDA_VISIBLE_DEVICES=3 ./CarlaUE4.sh --world-port=2006 -opengl  &
sleep 1
```

YilinGao-SHU, 2024-01-05:
Thanks so much! I will have a try!