opendilab/InterFuser #84: validation/evaluation loss

No4x, 2024-02-03:
>epoch,train_loss,eval_loss,eval_l1_error
0,0.13536748044323502,0,0
1,0.07556402581302743,0,0
2,0.055056504134023396,0,0
3,0.04527535825444941,0,0
4,0.03947842254450447,0,0
5,0.0358472646067017,0,0
6,0.03310371172663413,0,0
7,0.03174733430670019,0,0
8,0.03002995123204432,0,0
9,0.028877001517174535,0,0
10,0.02774458821666868,0,0
11,0.02721993143490532,0,0
12,0.02672633157487501,0,0
13,0.026068344869111713,0,0
14,0.02551674161171704,0,0
15,0.024949959142689118,0,0
16,0.024757716863563185,0,0
17,0.024379257975440275,0,0
18,0.023931162131198665,0,0
19,0.023686085660990917,0,0
20,0.023409886697405263,0,0
21,0.022983169294240182,0,0
22,0.022691530335629196,0,0
23,0.022611109690185177,0,0
24,0.022448889277221863,0,0
25,0.022408783370465563,0,0
26,0.022228354045696425,0,0
27,0.021992528582351248,0,0
28,0.02206064241758564,0,0
29,0.02181754263869503,0,0
30,0.021916808312137923,0,0
31,0.021604198315425923,0,0
32,0.021684262078059346,0,0
33,0.021629529524790612,0,0
34,0.02157683242439178,0,0
This is the summary.csv file, why is the eval_loss equal to 0? How could choose the best model? In the output folder the best model is the first checkpoint.

deepcs233, 2024-02-04:
Hi!
It looks like you didn't set the eval dataset correctly (the eval dataset may be empty).

About how to choose the best model, I usually choose the model that has the lowest eval loss or the model saved in the last epoch.

No4x, 2024-02-04:
Understood, thanks for your answer.

closed 2024-02-04 by No4x