opendilab/InterFuser #85: LeaderBoard evaluator object has no attribute manager

waleedrafique51, 2024-03-13:
The work present by author is excellent. I need help how to resolve this issue. Can author help me to resolve this issue for data collection .Here is error message  .please help me to sort out this bug
[image]

deepcs233, 2024-04-07:
Hi!
Sorry for the late reply. Could you check whether the Carla simulator ran normally? Or can you restart the simulator.

waleedrafique51, 2024-04-13:
Yes my carla is working properly,The issue is due to leaderboard script in learderboardevaluator.py.I have modified some lines of codes with the new leaderboard script .Now it is working

rogerfinnerty, 2024-05-27:
@waleedrafique51 I am getting the same issue, how did you fix this?

Ruyi001, 2024-07-15:
@waleedrafique51 I am getting the same issue, how did you fix this?