opendilab/InterFuser #89: Failed - Agent couldn't be set up

motianxiuhua, 2024-05-25:
hello, I am sorry to bother you, but I trully have trouble with the evaluation. Here is the problem:

I already set up the Carla,
[image]

and then set up the run_evaluation.sh, but it seems doesnot work
[image]

when I see the sample_result.json, it shows 

> Failed - Agent couldn't be set up

[image]

motianxiuhua, 2024-05-25:
[image]
And I have already installed timm

motianxiuhua, 2024-05-25:
I have already fixed it, just delete the sample_result.json generated at last time.

AlisonZhangyy, 2024-05-29:
maybe you should complied timm by youself, use the python file under folder interfuse named setup.py, as the readme said