opendilab/InterFuser #91: Could not set up the required agent: > Unknown model (interfuser_baseline)

ml6362662, 2024-05-27:
hello, I am sorry to bother you, but I trully have trouble with the evaluation. Here is the problem:

I already set up the Carla：
[image]
and then set up the run_evaluation.sh, but it seems doesnot work：
[image]
and And I have already installed timm (Use commands >>> pip timm ).Querying related issues, it seems to appear in timm, I guess it's a version issue, but I don't know how to fix it :
[image]
I'm a newbie to Carla and don't know how to fix this. If anyone knows how to fix this bug, I'd love to be able to tell me the details of how to fix it, which is crucial for me. Thank you so much!!!

zygalaxy, 2024-05-31:
Have you solved the problem? I have the same problem, still unresolved

zygalaxy, 2024-05-31:
I successfully resolved the issue by adding the following code at line 40 of the InterFuser/leaderboard/leaderboard/leaderboard_evaluator.py file:
[image]

ml6362662, 2024-06-04:
I successfully solved the problem: I deleted the downloaded project, re-downloaded and installed the project, and reconfigured the environment as required (found that the required installation would install timm properly, and there was no need to reinstall another version of timm, so I suspected another problem). For the model problem displayed, Found behind is not to download the weight of proprietary (Interfuser.pth.tar on Interfuser/leadervoard/team_code) and a compression package ((resnet50d_ra2-464 e36ba.pth) assessment script will automatically download, If the download fails, click the pop-up link to download it directly, and then put it in the corresponding folder (you will see the corresponding path when the download fails).）