opendilab/InterFuser #96: Issues when using interfuser agent with route_based scenario from scenario runner

TahaCarla, 2024-07-08:
Hello, hope you are doing well, thanks for your great work, I've been trying to use the Interfuser agent lately with the pretrained part, I could make the leaderboard evaluator work successfully, but unfortunately for some reason, using it as an agent with route-based scenarios is still not possible for me, I have been facing this issue related to sensors:

PS C:\Users\taha\Desktop\scenario_runner-0.9.10\scenario_runner-0.9.10> python scenario_runner.py --route C:\Users\taha\Desktop\scenario_runner-0.9.10\scenario_runner-0.9.10\srunner\data\routes_training.xml C:\Users\taha\Desktop\scenario_runner-0.9.10\scenario_runner-0.9.10\srunner\data\all_towns_traffic_scenarios.json 7 --agent C:\Users\taha\Desktop\InterFuser\InterFuser\leaderboard\team_code\interfuser_agent.py --agentConfig C:\Users\taha\Desktop\InterFuser\InterFuser\leaderboard\team_code\interfuser_config.py
pygame 2.1.2 (SDL 2.0.18, Python 3.7.9)
Hello from the pygame community. https://www.pygame.org/contribute.html
load model: C:/Users/taha/Desktop/InterFuser/InterFuser/leaderboard/team_code/interfuser.pth.tar
42routes_07_07_16_11_45
Preparing scenario: RouteScenario_7
Traceback (most recent call last):
  File "scenario_runner.py", line 398, in _load_and_run_scenario
    self.manager.load_scenario(scenario, self.agent_instance)
  File "C:\Users\taha\Desktop\scenario_runner-0.9.10\scenario_runner-0.9.10\srunner\scenariomanager\scenario_manager.py", line 112, in load_scenario
    self._agent.setup_sensors(self.ego_vehicles[0], self._debug_mode)
  File "C:\Users\taha\Desktop\scenario_runner-0.9.10\scenario_runner-0.9.10\srunner\autoagents\agent_wrapper.py", line 81, in setup_sensors
    sensor.listen(CallBack(sensor_spec['id'], sensor, self._agent.sensor_interface))
  File "C:\Users\taha\Desktop\scenario_runner-0.9.10\scenario_runner-0.9.10\srunner\autoagents\sensor_interface.py", line 31, in __init__
    self._data_provider.register_sensor(tag, sensor)
TypeError: register_sensor() missing 1 required positional argument: 'sensor'
register_sensor() missing 1 required positional argument: 'sensor'
No more scenarios .... Exiting

I also face this error when I try to run these scenarios with dummy agent, which is a strange one because dummy agent and agent wrapper should be consistent, but it has these errors because of sensors too.

 python scenario_runner.py --route C:\Users\taha\Desktop\scenario_runner-0.9.10\scenario_runner-0.9.10\srunner\data\routes_training.xml C:\Users\taha\Desktop\scenario_runner-0.9.10\scenario_runner-0.9.10\srunner\data\all_towns_traffic_scenarios.json 7 --agent C:\Users\taha\Desktop\scenario_runner-0.9.10\scenario_runner-0.9.10\srunner\autoagents\dummy_agent.py
Preparing scenario: RouteScenario_7
Traceback (most recent call last):
  File "scenario_runner.py", line 398, in _load_and_run_scenario
    self.manager.load_scenario(scenario, self.agent_instance)
  File "C:\Users\taha\Desktop\scenario_runner-0.9.10\scenario_runner-0.9.10\srunner\scenariomanager\scenario_manager.py", line 112, in load_scenario
    self._agent.setup_sensors(self.ego_vehicles[0], self._debug_mode)
  File "C:\Users\taha\Desktop\scenario_runner-0.9.10\scenario_runner-0.9.10\srunner\autoagents\agent_wrapper.py", line 61, in setup_sensors
    bp.set_attribute('range', str(sensor_spec['range']))
KeyError: 'range'
'range'
No more scenarios .... Exiting

I found in LAV agent that you should add vulkan flag and this also generates an issue for me, is it also needed to run interfuser or there is something i'm doing wrong that I don't know.
 Thank you in advance.

deepcs233, 2024-07-11:
Hi!

Could you provide more information about your setting and why did you need to run "python scenario_runner.py"? I'm not familiar with this entry script. The two problems look like the reason is the sensors' error set. The correct set can refer tohttps://github.com/opendilab/InterFuser/blob/4145d6ca58c19ce2a29d319c47980cea05586b06/leaderboard/team_code/interfuser_agent.py#L244 .If you need to specify the test routes or scenarios, modifying the environmental varieties in the evaluation script is just ok.

TahaCarla, 2024-07-11:
Hello! Thank you for your reply.

python scenario_runner.py 
--route C:\Users\taha\Desktop\InterFuser\InterFuser\leaderboard\data\training_routes\routes_town05_long.xml C:\Users\taha\Desktop\InterFuser\InterFuser\leaderboard\data\scenarios\town05_all_scenarios.json 16
 --agent C:\Users\taha\Desktop\InterFuser\InterFuser\leaderboard\team_code\interfuser_agent.py 
--agentConfig C:\Users\taha\Desktop\InterFuser\InterFuser\leaderboard\team_code\interfuser_config.py

In this command, I am running scenario runner in order to run a route-based scenario with interfuser as an agent like specified in the code inside scenario runner: 

parser.add_argument(
        '--route', help='Run a route as a scenario (input: (route_file,scenario_file,[route id]))', nargs='+', type=str)
parser.add_argument(
        '--agent', help="Agent used to execute the scenario. Currently only compatible with route-based scenarios.")    parser.add_argument('--agentConfig', type=str, help="Path to Agent's configuration file", default="")

This way, we could use autonomous_agent and human_agent from autoagents folder with those scenarios and it worked fine, but we had problems when agents need sensors even if we're using exactly the same files from the github link and we did change the environment variables to look for the scenario runner inside interfuser folder. If there is still something unclear about our situation, please feel free to ask.

Thank you!

deepcs233, 2024-07-14:
Hi!

I'm very sorry that I cannot solve your problem. The issue is not from the Interfuser itself and I have never used scenario_runner.py to evaluate the agent before. However, I have some suggestions which may help you:

1. Raise an issue in these repos: https://github.com/carla-simulator/carla https://github.com/carla-simulator/scenario_runner
2. Upgrade/ Degrade the version of scenario_runner
3. After running the agents with sensors successfully, you may try to use the Interfuser agent again.