opendilab/InterFuser #103: 夜晚天气数据

littleblackzi, 2024-10-28:
[image]
你好，我有几个问题想请教一下，方便起见，我直接用中文提问了
1.请问作者用auto_pilot收集到的数据中，有没有含夜晚的数据？有没有使用夜晚的数据用于训练？
2.我在查阅了Carla的官方文档和代码后，了解到作者自定义的7种夜晚天气，例如clearnight(5.0,0.0,0.0,10.0,-1.0,-90.0,60.0,75.0,1.0,0.0)参数的含义，其中，有关雾的参数应该是60.0,75.0,1.0,0.0中的其中三个，但我不能确定是哪三个，包括fog_density，fog_distance，fog_falloff。我的问题是，carla0.9.10.1是否具备渲染出雾的效果的能力呢？因为夜晚光线不是很好，我尝试过修改Carla中自定义的14种天气的参数，当然只是采用控制变量法改变他们的雾的参数，但是在采集数据的过程中，发现并没有出现雾的效果，只有天空是灰蒙蒙的，摄像头采集的图片中也没有出现雾，依旧是很清晰的图像。请问我如何才能采集到带有雾的数据呢？

TianYifan-Vincent, 2024-10-29:
我也在尝试采集雾的数据，我发现通过carla9.13运行自带的environment脚本是可以出现雾的，我怀疑是carla9.10运行OpenGL的原因，但我不知道该项目是否能适配carla9.13

TianYifan-Vincent, 2024-10-29:
你可以尝试使用-prefernvidia代替-opengl，在尝试不同的天气收集的图像，但是会出现噪声如图。不知道你有没有出现这种情况
[image]

littleblackzi, 2024-10-29:
> 你可以尝试使用-prefernvidia代替-opengl，在尝试不同的天气收集的图像，但是会出现噪声如图。不知道你有没有出现这种情况 [...]

你好，我知道如何在0.9.10.1中渲染出雾的效果了。
但是我也遇到了噪声问题，目前没有解决方法，或许跟所使用的显卡性能有关？

TianYifan-Vincent, 2024-10-30:
> 你好，我知道如何在0.9.10.1中渲染出雾的效果了。 但是我也遇到了噪声问题，目前没有解决方法，或许跟所使用的显卡性能有关？

应该和引擎有关吧，我看carla0.9.13里的画面没有出现这种噪声，我打算看能不能升级一下carla版本，有进展的话我们可以一起交流

TianYifan-Vincent, 2024-10-30:
[image]
[image]
如图是carla9.13拍摄的

littleblackzi, 2024-10-30:
> 我也在尝试采集雾的数据，我发现通过carla9.13运行自带的environment脚本是可以出现雾的，我怀疑是carla9.10运行OpenGL的原因，但我不知道该项目是否能适配carla9.13

理论上这个项目应该是可以迁移到0.9.13上的，但是需要一定的技术支持，除非你对本项目和Carla很熟悉，否则是有点难度的，这个我也不知道如何操作，需要修改哪些部分，很遗憾帮不到你。
另外，还有一种渲染方式，将-opengl替换成-vulkan试试，似乎-vulkan也会出现噪声，但没有-prefernvidia那么严重，你可以试试，然后对比一下（tip：-vulkan需要安装一下）

TianYifan-Vincent, 2024-10-30:
好的感谢你，我试一下vulkan

deepcs233, 2024-10-30:
@littleblackzi 包含了夜晚的数据，也使用了夜晚的数据。 生成有雾的场景我也确实不了解，升级carla版本可能需要调整leaderboard和scenario_runner的版本。

TianYifan-Vincent, 2024-10-30:
> > 我也在尝试采集雾的数据，我发现通过carla9.13运行自带的environment脚本是可以出现雾的，我怀疑是carla9.10运行OpenGL的原因，但我不知道该项目是否能适配carla9.13
> 
> 理论上这个项目应该是可以迁移到0.9.13上的，但是需要一定的技术支持，除非你对本项目和Carla很熟悉，否则是有点难度的，这个我也不知道如何操作，需要修改哪些部分，很遗憾帮不到你。 另外，还有一种渲染方式，将-opengl替换成-vulkan试试，似乎-vulkan也会出现噪声，但没有-prefernvidia那么严重，你可以试试，然后对比一下（tip：-vulkan需要安装一下）

您好，我尝试安装了vulkan，vulkaninfo也显示输出了，但是在运行`./carla/CarlaUE4.sh -vulkan`之后还是会报这个分段错误，您知道怎么解决吗，似乎我不加-opengl或-prefernvidia就会报分段错误。
```
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
X Error of failed request:  BadMatch (invalid parameter attributes)
  Major opcode of failed request:  149 ()
  Minor opcode of failed request:  4
  Serial number of failed request:  409
  Current serial number in output stream:  428
terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Signal 6 caught.
Segmentation fault (core dumped)

```

littleblackzi, 2024-10-30:
> > > 我也在尝试采集雾的数据，我发现通过carla9.13运行自带的environment脚本是可以出现雾的，我怀疑是carla9.10运行OpenGL的原因，但我不知道该项目是否能适配carla9.13
> > 
> > 
> > 理论上这个项目应该是可以迁移到0.9.13上的，但是需要一定的技术支持，除非你对本项目和Carla很熟悉，否则是有点难度的，这个我也不知道如何操作，需要修改哪些部分，很遗憾帮不到你。 另外，还有一种渲染方式，将-opengl替换成-vulkan试试，似乎-vulkan也会出现噪声，但没有-prefernvidia那么严重，你可以试试，然后对比一下（tip：-vulkan需要安装一下）
> 
> 您好，我尝试安装了vulkan，vulkaninfo也显示输出了，但是在运行`./carla/CarlaUE4.sh -vulkan`之后还是会报这个分段错误，您知道怎么解决吗，似乎我不加-opengl或-prefernvidia就会报分段错误。
> 
> ```
> 4.24.3-0+++UE4+Release-4.24 518 0
> Disabling core dumps.
> X Error of failed request:  BadMatch (invalid parameter attributes)
>   Major opcode of failed request:  149 ()
>   Minor opcode of failed request:  4
>   Serial number of failed request:  409
>   Current serial number in output stream:  428
> terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
> Signal 6 caught.
> Segmentation fault (core dumped)
> ```
> 
> 我采用`export VK_ICD_FILENAMES="/usr/share/vulkan/icd.d/nvidia_icd.json"之后成功了，但是拍出来的图像似乎没差 ![0009](https://private-user-images.githubusercontent.com/72920835/381467971-0d1739d5-430c-40c9-80d6-1a432eb69ef2.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAyOTQ1MzAsIm5iZiI6MTczMDI5NDIzMCwicGF0aCI6Ii83MjkyMDgzNS8zODE0Njc5NzEtMGQxNzM5ZDUtNDMwYy00MGM5LTgwZDYtMWE0MzJlYjY5ZWYyLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDMwVDEzMTcxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJkZjA5YzlkMzNhMjMzNGU1ZWQwNzkxN2FiZTE3Mjg3YWRlNDUxZmUxMmZlNDI5Y2U5OTgxZjY3Zjg0ZjEwNjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.DjgyKAuHRHMO4o-xkEirQW9AcqQvk_PqH6FnJzLsBKc)

我这里没有报你所说的分段错误。
至于噪声问题，我表示没什么解决的头绪，头一次遇到。