opendilab/InterFuser #107: Unknown model (interfuser_baseline) / CUDA errorv / timm

ZyhNoPanic, 2024-11-21:
I am a beginner in end-to-end autonomous driving and am trying to revive the InterFuser project.
My GPU is NVIDIA 4080 Super and it brings me some tough problems which hold me back.
I went through an extremely difficult experience and finally solved them, and now I wanna share it with you all.
If you find some problems in the solution or have better alternative please let me know. I'd appreciate it very much！

#### `sm89` CUDA Capability
At first My CUDA version is 11.4 and Pytorch version is 1.12.x, virtual env with python 3.7 (created by requirements.txt)
BUT NVIDIA 40 Series's CUDA Capability is `sm89`, and the Traceback suggests that it is not compatible with the current PyTorch.
So some error emerge
[image]
After some research it suggests that `sm89` needs CUDA 12.x
So I reinstall CUDA 12.1 and PyTorch which matches it, and find another problem.
#### PyTorch's oldest version which matches CUDA 12.x doesn't support Python 3.7
It at least need python 3.8
So I didn't listen to README.MD, created env with python 3.8
And this makes `requirements.txt`-installation meet some conflicts.
I made some adjustments by myself and try to run `run_evaluation.sh`
And I suspect this is why my `timm` module was not correctly installed.
#### No module named 'timm'
This is the third problem, when I run `run_evaluation.sh`
I first `pip install timm`, and clearly it didn't work.
Then I find some clue in Issues and find timm version was 0.4.13.
BUT `pip install timm==0.4.13` didn't work too.
But there is a floder a folder in the project `InterFuser/interfuser/timm`, Yes, the floder named `timm`
Then I went to my Anaconda3 floder and fine the floder with the same name, finding them having same structre.
A Serendipitous discovery after my three days trying.
So I tried to replace the 'timm' floder in Anaconda3 with the one in `InterFuser/interfuser/timm`
And this time it worked！！！
#### BTW, if running `run_evaluation.sh` with this result
[image]
just delete the file in `InterFuser/results/sample_result.json`
## Hope that my experience can provide you with some references : )

deepcs233, 2024-11-25:
Hi!

Thank you so much for your kind help! I’ve added the "good first issue" tag to attract more attention from newcomers and updated the README with the corresponding link. By the way, regarding the installation of timm, we’ve provided detailed instructions in the "Setup" section.

ZyhNoPanic, 2024-11-27:
Thanks for reminding！For some reason, the installation was not successful at the beginning, and I carelessly ignored it later. I've modified origin issue.