opendilab/InterFuser #110: Some training losses are on the rise

ZyhNoPanic, 2025-02-28:
Hello, I'm really appriciate your excellent work and great code.
Now as an E2E beginner I'm trying to reproduce this work on my computer in a smaller scale.
I tried to train my own model in a very small scale data:
With data generated by `routes_town01/02_long/short/tiny_w0`,only town01 and 02 in weather-0, about 46G data.
And all hyper-parameters were set by the default value of InterFuser repo
After about 35 epochs
The **Loss Graph** looks like
[image]
##### Why loss of traffic_light_state, junction and stop_sign went up?
And I have tested the model I trained in town03, itt performed the same as the loss graph suggesting, the detection of traffic light, stop_sign and junction is awful while predicting velocity and direction very well. 
It's really confusing.
Was it caused by my parameters settings or small scale data or few training epochs?
#### On the other hand
The perception visualizing maps on the right side was not accurate. How was that part trained. Was it trained by the BEVmap generated by Carla?, And with more training, it will be better right?

Thank you so much!

deepcs233, 2025-02-28:
1. Town01/02's maps only include single-lane roads, which have a huge gap with Town03.
2. The loss of traffic_light_state, junction and stop_sign went up, but you can find their accuracy still became better. It's because of the CE loss's over-confidence and it's a common phenomenon that you can find more details in the Internet.  
3. The perception visualizing maps was trained with the information (size, position, etc,.) of the surrounding objects. I think with more data and more training, its performance will become better.

ZyhNoPanic, 2025-03-01:
Thanks a lot ! :)

I'll try it again wirh more training and data.