opendilab/InterFuser #117: reproduct the interfuser performance result

fuyz1231, 2025-07-25:
Hi all,
Has anyone successfully reproduced the InterFuser results?
I’ve collected long routes for Towns 1–5 and short routes for Towns 7 and 10, across all 14 weather conditions. However, after training, the driving score is still around 50.
Any suggestions or thoughts?

[image]

[image]

fuyz1231, 2025-07-25:
wrong image. Here is the right one

[image]

deepcs233, 2025-08-07:
How about your datset size? long/short/tiny data should all be collected. You can also consider the provided dataset used in the LMdrive.